mne-tools/mne-python #11895: Possible (small) errors in ERDS maps example

cbrnr, 2023-08-17:
### Proposed documentation enhancement

I just noticed the following two issues in our [ERDS maps example](https://mne.tools/stable/auto_examples/time_frequency/time_frequency_erds.html):

1. The last two plots include "ERDS (%)" as their y axes labels. I think this is not correct, since the ERDS values are not in %. There are two options: we can either convert ERDS values to %, or drop the "(%)" from the labels.
2. I am not sure why the last plot shows negative values up to about -2, because the minimum possible ERD value is -1 (-100%, meaning power during activity completely disappeared as compared to baseline). Is this a visualization artifact of the violin plots?

@eioe do you have time to take a look?

eioe, 2023-08-17:
Hi @cbrnr , 
thanks for spotting these. I will try to take a look tonight or tomorrow. 
If I'm too slow to get to a satisfying solution, I'd write here again as I'm then off for 4 weeks of vacation (starting tomorrow evening).

eioe, 2023-08-19:
@cbrnr , thanks a lot for the clarification. I read that in the docs, but did not pay enough attention to realize that we're missing the multiplication with 100. Indeed, a bit misleading naming. 
Re. your first point: any of the two mentioned options sounds reasonable to me; with a slight preference for multiplying the values with 100 and keeping the "%" in the label, so that we have some kind of a unit on the axis label. What do you think?

Re. 2: Setting the [parameter `cut=0` in the call  of `sns.violinplot`](https://seaborn.pydata.org/generated/seaborn.violinplot.html) should yield the wanted behavior (e.g., in [line 210 here](https://github.com/mne-tools/mne-python/blob/6fe2e1322afb6ae5f6816f05d8f85dd78678e1aa/examples/time_frequency/time_frequency_erds.py#L210)). This is how it looks:

[image]

unfortunately, I did not find enough time today to set up a mne-dev environment on this machine here, so i could create a PR. And we are about to leave for our vacation. 
If it is ok for yout, I'm thankful if you go ahead and fix/adapt it. Or – if you prefer – I am happy to do it once I'm back, but that will only be mid september. Sorry, unfortunate timing.

Thanks anyways & very cool that you spotted this. 🙌

cbrnr, 2023-08-19:
Thanks @eioe, I've submitted a PR with the changes. I didn't convert the values to %, because this would have been more work (because I'd also need to change the t/f plots), so I've just fixed the y-axis labels. Have a great vacation!

PR #11904 (2023-08-21): Minor fixes for ERDS maps example
Fixes #11895. This PR corrects the y-axis labels (from "ERDS (%)" to just "ERDS") and changes the violin plots so that violins do not exceed the data range (-1 is the lowest possible value).
closed 2023-08-21 by agramfort