mne-tools/mne-python #11901: update rank estimation in compute_source_psd()

olafhauk, 2023-08-18:
### Description of the problem

mne.minimum_norm.compute_source_psd() still uses "old-fashined" rank estimation and may have problems with maxfiltered data. 
For example, the part

```
if pca:
        U, s, Vh = linalg.svd(K, full_matrices=False)
        rank = np.sum(s > 1e-8 * s[0])
        K = s[:rank] * U[:, :rank]
        Vh = Vh[:rank]
        logger.info("Reducing data rank %d -> %d" % (len(s), rank))
    else:
        Vh = None
    is_free_ori = inverse_operator["source_ori"] == FIFF.FIFFV_MNE_FREE_ORI
```
should use the estimated rank of the `data.`

### Steps to reproduce

```Python
I used it like

stc_psd, evo_psd = mne.minimum_norm.compute_source_psd(
raw=raw, inverse_operator=invop, lambda2=1 / 9., method='MNE',
fmin=0., fmax=140., n_fft=60000, overlap=0.5, pca=True,
nave=1, bandwidth='hann', low_bias=True, return_sensor=True)
```

and for 6 our of my 29 subjects it produced

Reducing data rank 370 → 1

and indeed in the output all times courses (source space and sensor space) were the same.
```

### Expected results

It should have used the rank after maxfilter (for MEG). 

### Actual results

Output include (for 6 out of 29 subjects)

Reducing data rank 370 → 1

and all time courses (source and sensor space) indeed are the same.

### Additional information

Platform             Linux-3.10.0-1160.el7.x86_64-x86_64-with-glibc2.17
Python               3.10.10 | packaged by conda-forge | (main, Mar 24 2023, 20:08:06) [GCC 11.3.0]
Executable           /imaging/local/software/mne_python/mne1.4.0_1/bin/python3.10
CPU                  x86_64 (16 cores)
Memory               188.7 GB

Core
?? mne               1.4.0
?? numpy             1.21.6 (MKL 2022.2-Product with 16 threads)
?? scipy             1.10.1
?? matplotlib        3.7.1 (backend=QtAgg)
?? pooch             1.7.0
?? jinja2            3.1.2

Numerical (optional)
?? sklearn           1.2.2
?? numba             0.56.4
?? nibabel           5.1.0
?? nilearn           0.10.1
?? dipy              1.7.0
?? openmeeg          2.5.6
?? pandas            2.0.1
?? unavailable       cupy

Visualization (optional)
?? pyvista           0.39.0 (OpenGL 3.3 (Core Profile) Mesa 18.3.6 via llvmpipe (LLVM 3.4, 256 bits))
?? pyvistaqt         0.0.0
?? ipyvtklink        0.2.2
?? vtk               9.2.6
?? qtpy              2.3.1 (PyQt5=5.15.6)
?? ipympl            0.9.3
?? pyqtgraph         0.13.3
?? mne-qt-browser    0.5.0

Ecosystem (optional)
?? mne-bids          0.12
?? mne-nirs          0.5.0
?? mne-features      0.2.1
?? mne-connectivity  0.5.0
?? mne-icalabel      0.4
?? mne-bids-pipeline 1.2.0

agramfort, 2023-08-21:
can you share a file and a script privately to replicate the problem?

Message ID: ***@***.***>
>

olafhauk, 2023-08-21:
I've just shared data and script via Google Drive.

agramfort, 2023-08-27:
@olafhauk it appears to be an issue with your inverse operator. Can you share the files to replicate the computation of the inverse operator? I suspect an issue with the noise covariance you used.

olafhauk, 2023-08-30:
I think I found the problem. I computed the covariance matrices on "rest" blocks and applied them to data from "face" blocks. For some subjects the rank of the MEG data for the "rest" blocks was different than for the "face" blocks (presumably by Maxfilter). For the Inverse Operator, I used the 'info' from the "face" blocks. If I change this to the 'info' from "rest" blocks it seems to be fine.
Now I'm asking myself what to do if the rank after Maxfilter is different for different blocks for the same subject. Will MNE use the correct rank?
I uploaded new data to the Google Drive, in case you want to have a look. The main files are 'rest1_sss_f_ica_raw.fif' and 'face_sss_f_ica_raw.fif'.

Thank you,

Olaf

larsoner, 2023-08-30:
> Now I'm asking myself what to do if the rank after Maxfilter is different for different blocks for the same subject. Will MNE use the correct rank?

Not necessarily, you have to be mindful and set `rank` and use diagnostic plots etc. :( See previous discussions like https://github.com/mne-tools/mne-python/issues/1902, https://github.com/mne-tools/mne-python/issues/4676, https://github.com/mne-tools/mne-python/issues/11539, https://github.com/mne-tools/mne-python/pull/8805, https://github.com/mne-tools/mne-bids-pipeline/issues/760, etc. I think we've discussed it elsewhere, too, but that's what I found with a quick search.

olafhauk, 2023-08-30:
Thanks - I'll now be away for a few days but will get back to it next week.