mne-tools/mne-python #11908: warnings in sys_info() output

drammock, 2023-08-21:
This showed up in a user's `mne.sys_info()` output [on the forum](https://mne.discourse.group/t/valueerror-all-picks-must-be-n-channels-73-got-73/7375/35).
```
Found Intel OpenMP ('libiomp') and LLVM OpenMP ('libomp') loaded at
the same time. Both libraries are known to be incompatible and this
can cause random crashes or deadlocks on Linux when loaded in the
same Python program.
Using threadpoolctl may cause crashes or deadlocks. For more
information and possible workarounds, please see
    https://github.com/joblib/threadpoolctl/blob/master/multiple_openmp.md

  warnings.warn(msg, RuntimeWarning)
 (OpenGL 4.5.0 NVIDIA 536.23 via NVIDIA GeForce RTX 2080 SUPER/PCIe/SSE2)
├☑ pyvistaqt         0.0.0
```

Seems worth digging into how this happened, and trying to prevent it.

larsoner, 2023-08-22:
OpenMEEG will use OpenBLAS (it compiles with / links to it), and I'm guessing the user installed numpy/scipy with the MKL BLAS backend

larsoner, 2023-08-22:
... this sort of conflict is difficult to guard against. Ideally OpenMEEG would be linalg-agnostic but that would be *a lot* of really annoying/difficult work

drammock, 2023-08-22:
if we can't prevent it, can we anticipate the error, catch the warning, and spit out something *at the end* of the `sys_info()` output?  (basically, make it prettier)

larsoner, 2023-08-22:
I doubt it (it's going to be emitted at the level of the C bindings I think) but you could try

hoechenberger, 2023-08-23:
> OpenMEEG will use OpenBLAS (it compiles with / links to it), and I'm guessing the user installed numpy/scipy with the MKL BLAS backend

Considering the path, it looks to me like this is an installation produced by our standalone installers, no? So it may be something for us to fix in the installer package specification

hoechenberger, 2023-08-23:
@larsoner According to https://github.com/mne-tools/mne-installers/releases/download/v1.5.0/MNE-Python-1.5.0_0-Windows.exe.env.json we ship libraries for MKL and OpenBLAS:

```json
  {
    "base_url": "https://conda.anaconda.org/conda-forge",
    "build_number": 17,
    "build_string": "17_win64_mkl",
    "channel": "conda-forge",
    "dist_name": "libblas-3.9.0-17_win64_mkl",
    "name": "libblas",
    "platform": "win-64",
    "version": "3.9.0"
  },
  ...
  {
    "base_url": "https://conda.anaconda.org/conda-forge",
    "build_number": 0,
    "build_string": "pthreads_hc140b1d_0",
    "channel": "conda-forge",
    "dist_name": "libopenblas-0.3.23-pthreads_hc140b1d_0",
    "name": "libopenblas",
    "platform": "win-64",
    "version": "0.3.23"
  },
  ...
  {
    "base_url": "https://conda.anaconda.org/conda-forge",
    "build_number": 0,
    "build_string": "pthreads_h1956dd5_0",
    "channel": "conda-forge",
    "dist_name": "openblas-0.3.23-pthreads_h1956dd5_0",
    "name": "openblas",
    "platform": "win-64",
    "version": "0.3.23"
  },
```
Maybe explicitly requesting `"libblas=*=*openblas"` could avoid this, see
https://conda-forge.org/docs/maintainer/knowledge_base.html#switching-blas-implementation

hoechenberger, 2023-08-23:
But are you sure this is the cause of the error message here? Because the message itself mentions OpenMP, not BLAS / MKL. Are they linked to one another?

hoechenberger, 2023-08-23:
@larsoner Could you transfer this issue to mne-tools/mne-installers please?

larsoner, 2023-08-23:
@hoechenberger the issue says "Intel OpenMP ('libiomp') and LLVM OpenMP ('libomp')" -- I think the former will be used by MKL (being Intel's "thing"), the latter by OpenBLAS (maybe Numba/LLVM as well)