mne-tools/mne-python #11931: Test failure for 1.5.0 on Fedora:

sanjayankur31, 2023-08-29:
### Description of the problem

A test fails when building mne-python from source on Fedora

### Steps to reproduce

```Python
The complete spec file is here:

https://src.fedoraproject.org/fork/ankursinha/rpms/python-mne/blob/b5469bbe631fb57fc09fbb4a453615e1b7ba9c1c/f/python-mne.spec#_140

We're just running the tests during the build like this:

export SETUPTOOLS_SCM_PRETEND_VERSION='%{version}'
export MNE_SKIP_TESTING_DATASET_TESTS=true
export MNE_SKIP_NETWORK_TESTS=1
export MNE_DONTWRITE_HOME=true
export MNE_FORCE_SERIAL=true
 
# Deselected tests require additional data or don't work in mock
# Two deselected for sklearn warnings
# Tools directory ignored as it contains tests for upstream release process
 
# required for some tests
mkdir subjects
 
# https://github.com/mne-tools/mne-python/blob/v1.0.3/tools/github_actions_test.sh#L7
# skip tests that require network
pytest -v -m "not (slowtest or pgtest)" \
    --deselect mne/datasets/tests/test_datasets.py \
    --deselect mne/utils/tests/test_numerics.py
 ```
```

### Expected results

All tests pass.

### Actual results

````
=================================== FAILURES ===================================
________________ test_cluster_permutation_with_adjacency[NumPy] ________________
mne/stats/tests/test_cluster_level.py:472: in test_cluster_permutation_with_adjacency
    assert len(w) == 1
E   assert 2 == 1
E    +  where 2 = len([<warnings.WarningMessage object at 0x7f2d3ee7c110>, <warnings.WarningMessage object at 0x7f2d3ee7cd10>])
```

### Additional information

The complete build log is attached:

[build.log.txt](https://github.com/mne-tools/mne-python/files/12466211/build.log.txt)

drammock, 2023-08-29:
thanks for the report, we'll look into it.

larsoner, 2023-08-30:
Feel free to try the branch from https://github.com/mne-tools/mne-python/pull/11934