mne-tools/mne-python #11947: BUG: info["subject_info"] not set when loading EDF

paulroujansky, 2023-09-01:
### Description of the problem

When loading an EDF file, the subject info located in the headers is set in `RawEDF._raw_extras` but not in `RawEDF.info`, causing the information to get lost when exporting the data to another EDF file (after adding annotations for instance).

This can be checked in `test_double_export_edf()` from [mne/export/tests/test_export.py#L119](https://github.com/mne-tools/mne-python/blob/maint/1.5/mne/export/tests/test_export.py#L119) by printing `info["subject_info"]` from both `raw` and `raw_read` objects.

The problem seems to come from the fact that subject info is stored in `RawEDF._raw_extras` but never actually set in `RawEDF.info`. This could be fixed in `get_info()` from [mne/io/edf/edf.py#L521](https://github.com/mne-tools/mne-python/blob/maint/1.5/mne/io/edf/edf.py#L521).

### Steps to reproduce

```Python
import mne

fname = "SC4001E0-PSG.edf"  # provide your path here
output_fname = "new_file.edf"

raw = mne.io.read_raw_edf(fname, preload=True)

print(raw._raw_extras[0]["subject_info"])

raw.export(output_fname)

raw2 = mne.io.read_raw_edf(output_fname, preload=True)

print(raw2._raw_extras[0]["subject_info"])
```

### Link to data

https://www.physionet.org/content/sleep-edfx/1.0.0/sleep-cassette/SC4001E0-PSG.edf

### Expected results

```
{'id': 'X', 'sex': 'F', 'birthday': 'X', 'name': 'Female_33yr'}
{'id': 'X', 'sex': 'F', 'birthday': 'X', 'name': 'Female_33yr'}
```

### Actual results

```
{'id': 'X', 'sex': 'F', 'birthday': 'X', 'name': 'Female_33yr'}
{'id': 'X', 'sex': 'X', 'birthday': 'X', 'name': 'X'}
```

### Additional information

The above code uses the latest release (`mne==1.5.0`).
The MWE uses an EDF file from Physionet's Sleep-EDF Database Expanded 1.0.0 as MNE's example EDF files do not seem to contain any patient information, neither does [mne/io/edf/tests/data/test.edf](https://github.com/mne-tools/mne-python/blob/maint/1.5/mne/io/edf/tests/data/test.edf).
A synthetic example could be built as well and used in tests. My suggestion would be to add subject info in [mne/io/edf/tests/data/test.edf](https://github.com/mne-tools/mne-python/blob/maint/1.5/mne/io/edf/tests/data/test.edf) and augment [mne/export/tests/test_export.py](https://github.com/mne-tools/mne-python/blob/maint/1.5/mne/export/tests/test_export.py)

larsoner, 2023-09-01:
Can you check if any of these files have the relevant info? We also test against them:

https://github.com/mne-tools/mne-testing-data/tree/master/EDF

As for what the key/value pairs should be we can use things that are in `subject_info` in the Notes section from here:

https://mne.tools/stable/generated/mne.Info.html

@paulroujansky would you be up for making a PR to fix the behavior on your files?

paulroujansky, 2023-09-02:
@larsoner I will look into that
Sure thing 👍

paulroujansky, 2023-09-04:
@larsoner I tried to load the files from https://github.com/mne-tools/mne-testing-data/tree/master/EDF with `mne.read_raw_edf()` but they are all raise the same error:
```
invalid literal for int() with base 10: 'nes="fal'
```

I tried to open them in EDFbrowser but I get the following error equally:
[image]

It seems like they are all corrupted or something.

cbrnr, 2023-09-04:
These files work fine for me. How did you download them?

paulroujansky, 2023-09-04:
Manually, that might be why 🙃. They work fine when I load them with `mne.datasets.testing`.

The files have the following subject info:
```
File: chtypes_edf.edf
-> raw._raw_extras[0]["subject_info"]:
{'id': '0', 'sex': 'X', 'birthday': datetime.datetime(1985, 6, 25, 0, 0), 'name': 'No_Name'}

File: SC4001EC-Hypnogram.edf
-> raw._raw_extras[0]["subject_info"]:
{'id': 'X', 'sex': 'F', 'birthday': 'X', 'name': 'Female_33yr'}

File: subsecond_starttime.edf
-> raw._raw_extras[0]["subject_info"]:
{'id': 'X', 'sex': 'F', 'birthday': datetime.datetime(1998, 1, 20, 0, 0), 'name': 'X,X'}

File: test_edf_overlapping_annotations.edf
-> raw._raw_extras[0]["subject_info"]:
{'id': 'X', 'sex': 'X', 'birthday': 'X', 'name': 'X'}

File: test_edf_stim_resamp.edf
-> raw._raw_extras[0]["subject_info"]:
{'id': 'X', 'sex': 'M', 'birthday': datetime.datetime(1967, 10, 9, 0, 0), 'name': 'X'}

File: test_generator_2.edf
-> raw._raw_extras[0]["subject_info"]:
{'id': 'X', 'sex': 'X', 'birthday': 'X', 'name': 'X'}

File: test_reduced.edf
-> raw._raw_extras[0]["subject_info"]:
{'id': 'X', 'sex': 'X', 'birthday': 'X', 'name': 'X'}

File: test_utf8_annotations.edf
-> raw._raw_extras[0]["subject_info"]:
{'id': 'X', 'sex': 'X', 'birthday': 'X', 'name': 'X'}
```

cbrnr, 2023-09-04:
> Manually, that might be why 🙃. They work fine when I load them with `mne.datasets.testing`

They work for me even when I download them manually (by clicking on the "Download raw file" icon). Maybe a browser thing?

closed 2023-09-05 by larsoner