mne-tools/mne-python #11959: BUG: annotation channels not saved when exporting to EDF file

paulroujansky, 2023-09-06:
### Description of the problem

When defining `ch_names` for annotations, the information gets lost when exporting the data to an EDF file, which is problematic when dealing with channel-specific events.

As defined in https://www.edfplus.info/specs/edftexts.html _3.3. Linking annotations to signal channels_ section, the channel information can be saved by adding `@@{ch_name}` to the description of each corresponding annotation.

### Steps to reproduce

```Python
import mne

fpath = "test.edf"
output_fname = r"test2.edf"

raw = mne.io.read_raw_edf(fpath)

evts = [
    {"onset": 0.1, "duration": 0.2, "description": "A", "ch_names": ["F1"]},
    {"onset": 0.5, "duration": 0.7, "description": "A", "ch_names": ["F2", "F1"]},
    {"onset": 2.0, "duration": 0.2, "description": "B", "ch_names": ["F1"]},
    {"onset": 3.2, "duration": 0.3, "description": "A", "ch_names": []},
]

onset = [evt["onset"] for evt in evts]
duration = [evt["duration"] for evt in evts]
description = [evt["description"] for evt in evts]
ch_names = [evt["ch_names"] for evt in evts]

orig_time = raw.info["meas_date"]

ann = mne.Annotations(onset, duration, description, orig_time, ch_names)

raw.set_annotations(ann)

mne.export.export_raw(output_fname, raw, overwrite=True)

raw2 = mne.io.read_raw_edf(output_fname)

print(raw.annotations.to_data_frame(), end="\n\n")
print(raw2.annotations.to_data_frame())
```

### Link to data

https://github.com/mne-tools/mne-python/blob/main/mne/io/edf/tests/data/test.edf

### Expected results

```
                    onset  duration description  ch_names
0 2014-04-29 22:19:44.100       0.2           A     (F1,)
1 2014-04-29 22:19:44.500       0.7           A  (F2, F1)
2 2014-04-29 22:19:46.000       0.2           B     (F1,)
3 2014-04-29 22:19:47.200       0.3           A        ()

                    onset  duration description
0 2014-04-29 22:19:44.100       0.2           A     (F1,)
1 2014-04-29 22:19:44.500       0.7           A  (F2, F1)
2 2014-04-29 22:19:46.000       0.2           B     (F1,)
3 2014-04-29 22:19:47.200       0.3           A        ()
```

### Actual results

```
                    onset  duration description  ch_names
0 2014-04-29 22:19:44.100       0.2           A     (F1,)
1 2014-04-29 22:19:44.500       0.7           A  (F2, F1)
2 2014-04-29 22:19:46.000       0.2           B     (F1,)
3 2014-04-29 22:19:47.200       0.3           A        ()

                    onset  duration description
0 2014-04-29 22:19:44.100       0.2           A
1 2014-04-29 22:19:44.500       0.7           A
2 2014-04-29 22:19:46.000       0.2           B
3 2014-04-29 22:19:47.200       0.3           A
```

### Additional information

The above code uses the latest release (`mne==1.5.0`).

paulroujansky, 2023-09-06:
I wrote some minimal code that fixes the issue. I'd be happy to create a PR

cbrnr, 2023-09-06:
Thank you @paulroujansky, it would be great if you submitted a PR to fix this issue!

closed 2023-09-06 by larsoner