mne-tools/mne-python #11961: Read raw eeglab bug

MAX00008888, 2023-09-07:
### Description of the problem

raw = mne.io.read_raw_eeglab('f001.set')

Traceback (most recent call last):
  File "C:\Users\forever\PycharmProjects\eeg\ceshishi.py", line 17, in <module>
    raw = mne.io.read_raw_eeglab('f001.set')
  File "C:\Users\forever\PycharmProjects\eeg\venv\lib\site-packages\mne\io\eeglab\eeglab.py", line 259, in read_raw_eeglab
    return RawEEGLAB(input_fname=input_fname, preload=preload,
  File "<decorator-gen-277>", line 12, in __init__
  File "C:\Users\forever\PycharmProjects\eeg\venv\lib\site-packages\mne\io\eeglab\eeglab.py", line 358, in __init__
    eeg = _check_load_mat(input_fname, uint16_codec)
  File "C:\Users\forever\PycharmProjects\eeg\venv\lib\site-packages\mne\io\eeglab\eeglab.py", line 60, in _check_load_mat
    eeg = _readmat(fname, uint16_codec=uint16_codec)
  File "C:\Users\forever\PycharmProjects\eeg\venv\lib\site-packages\mne\io\eeglab\_eeglab.py", line 82, in _readmat
    return read_mat(fname, uint16_codec=uint16_codec)
  File "C:\Users\forever\PycharmProjects\eeg\venv\lib\site-packages\pymatreader\pymatreader.py", line 92, in read_mat
    raw_data = scipy.io.loadmat(fid, struct_as_record=True,
  File "C:\Users\forever\PycharmProjects\eeg\venv\lib\site-packages\scipy\io\matlab\_mio.py", line 226, in loadmat
    matfile_dict = MR.get_variables(variable_names)
  File "C:\Users\forever\PycharmProjects\eeg\venv\lib\site-packages\scipy\io\matlab\_mio5.py", line 332, in get_variables
    res = self.read_var_array(hdr, process)
  File "C:\Users\forever\PycharmProjects\eeg\venv\lib\site-packages\scipy\io\matlab\_mio5.py", line 292, in read_var_array
    return self._matrix_reader.array_from_header(header, process)
  File "_mio5_utils.pyx", line 671, in scipy.io.matlab._mio5_utils.VarReader5.array_from_header
  File "_mio5_utils.pyx", line 719, in scipy.io.matlab._mio5_utils.VarReader5.array_from_header
  File "_mio5_utils.pyx", line 963, in scipy.io.matlab._mio5_utils.VarReader5.read_struct
  File "_mio5_utils.pyx", line 669, in scipy.io.matlab._mio5_utils.VarReader5.read_mi_matrix
  File "_mio5_utils.pyx", line 713, in scipy.io.matlab._mio5_utils.VarReader5.array_from_header
  File "_mio5_utils.pyx", line 876, in scipy.io.matlab._mio5_utils.VarReader5.read_char
TypeError: buffer is too small for requested array

### Steps to reproduce

```Python
1
```

### Expected results

1

### Actual results

1

### Additional information

1

MAX00008888, 2023-09-07:
Hello, I use mne.io.read_raw_eeglab to read .set flie, but it shows TypeError: buffer is too small for requested array

MAX00008888, 2023-09-07:
A few months ago, when I ran the code, it was successful. I don't what' wrong with it. Could you please help me solve this?

larsoner, 2023-09-07:
I strongly suspect that your file got corrupted somehow. Can you read any other files?

A much less likely possibility is that you updated SciPy and their matlab reading code broke.

MAX00008888, 2023-09-08:
> I strongly suspect that your file got corrupted somehow. Can you read any other files?
> 
> A much less likely possibility is that you updated SciPy and their matlab reading code broke.

I can read former files, but I can't read new created files in EEGLAB

MAX00008888, 2023-09-08:
Do I need to resatll Matalab or SciPy？

MAX00008888, 2023-09-08:
Do I need to reinsatll Matalab or SciPy？

MAX00008888, 2023-09-08:
Do I need to reinstall Matalab or SciPy？

larsoner, 2023-09-08:
If they are new files they actually shouldn't be routed to `scipy.io` but should use `h5py` in `pymatreader` directly.

Can you create a tiny problematic file and share it? Then we can figure out if this is indeed a problem with `pymatreader` or something else.

MAX00008888, 2023-09-08:
[flie.zip](https://github.com/mne-tools/mne-python/files/12555439/flie.zip)
Of course, Sir. THANKS A LOT

MAX00008888, 2023-09-08:
> [flie.zip](https://github.com/mne-tools/mne-python/files/12555439/flie.zip) Of course, Sir. THANKS A LOT

This is the .set flie created in EEGLAB

larsoner, 2023-09-08:
Looks like this is a SciPy bug since `load 1.set` works in octave, and it's not a HDF5 file -- in `octave` I see:
```
> type 1.set
MATLAB 5.0 MAT-file, Platform: PCWIN64, Created on: Fri Sep  8 11:34:03 2023
...
```
Can you open a bug report [there](https://github.com/scipy/scipy/issues/new/choose) instead?