mne-tools/mne-python #11966: BUG: `raw.anonymize()` not fully anonymizing

paulroujansky, 2023-09-07:
### Description of the problem

When anonymizing a `raw` object with func `anonymize()`, the subject info and `meas_date` are still accessible through `raw._raw_extras` attribute as shown in the below example.

Regarding subject info, it would be easier in my opinion to remove field "subject_info" from both `raw.info` and `*raw._raw_extras` (or simply empty unecessary fields, keeping "his_id" only for instance) as it is not expected to be able to retrieve the information afterwards by definition.

### Steps to reproduce

```Python
import mne

data_path = mne.datasets.testing.data_path(download=False)
fpath = data_path / "EDF" / "test_edf_stim_resamp.edf"

raw = mne.io.read_raw_edf(fpath, verbose=False)

print("Subject info before anonymization")
print("\t", raw.info["subject_info"])
print("\t", raw.info["meas_date"])
print("\t", raw._raw_extras[0]["subject_info"])
print("\t", raw._raw_extras[0]["meas_date"])

raw.anonymize()

print("\t", "\nSubject info after anonymization")
print("\t", raw.info["subject_info"])
print("\t", raw.info["meas_date"])
print("\t", raw._raw_extras[0]["subject_info"])
print("\t", raw._raw_extras[0]["meas_date"])
print(raw._raw_extras[0]["subject_info"])
```

### Expected results

```
Subject info before anonymization
         {'his_id': 'X', 'sex': 1, 'last_name': 'X', 'birthday': (1967, 10, 9)}
         2016-02-14 10:33:35+00:00
         {'id': 'X', 'sex': 'M', 'birthday': datetime.datetime(1967, 10, 9, 0, 0), 'name': 'X'}
         2016-02-14 10:33:35+00:00

Subject info after anonymization
         {'his_id': '0', 'sex': 0, 'last_name': 'mne_anonymize', 'birthday': (1951, 8, 25)}
         2000-01-01 00:00:00+00:00
         {'id': '0', 'sex': 'unknown', 'birthday': datetime.datetime(1951, 8, 25, 0, 0), 'name': 'mne_anonymize'}
         2000-01-01 00:00:00+00:00

### Actual results

```
Subject info before anonymization
         {'his_id': 'X', 'sex': 1, 'last_name': 'X', 'birthday': (1967, 10, 9)}
         2016-02-14 10:33:35+00:00
         {'id': 'X', 'sex': 'M', 'birthday': datetime.datetime(1967, 10, 9, 0, 0), 'name': 'X'}
         2016-02-14 10:33:35+00:00

Subject info after anonymization
         {'his_id': '0', 'sex': 0, 'last_name': 'mne_anonymize', 'birthday': (1951, 8, 25)}
         2000-01-01 00:00:00+00:00
         {'id': 'X', 'sex': 'M', 'birthday': datetime.datetime(1967, 10, 9, 0, 0), 'name': 'X'}
         2016-02-14 10:33:35+00:00
```

### Additional information

The above code uses the latest release (`mne==1.5.1`).

paulroujansky, 2023-09-07:
Indeed, I was actually concerned about some compliance issues one might have by anonymizing _in-memory_ in his own app for instance, and therefore potentially exposing the unanomyzed data to malicious users, but it's definitely an edge case and would not be recommended anyhow.

larsoner, 2023-09-07:
> I was actually concerned about some compliance issues one might have by anonymizing in-memory in his own app for instance, and therefore potentially exposing the unanomyzed data to malicious users, but it's definitely an edge case

If it is an important use case for you, then that's at least one confirmed use :)

Based on your activity, I'm guessing you're working with (just) EDF for now, is that right? 😄  If so, maybe the thing to do is open a PR to keep any non-anonymized keys out of `_raw_extras` entirely before/independent of `.anonymize()`, i.e., in the `read_raw_edf` `_raw_extras` setting process. In theory, non-anonymized info shouldn't be necessary to facilitate reading the data (which is the purpose of `_raw_extras`). A PR for this would be welcome -- I consider less "stuff" hanging around in `_raw_extras` a win independent of the anonymization consideration (to some extent anyway)!

As part of that PR, we could document somewhere -- probably `mne/io/base.py` where we talk about what subclasses of `BaseRaw` *should* do (like implement `_read_segment_file`) -- that it's best practice to avoid any non-anonymized data in `_raw_extras`.

Then in follow-up PRs we could proceed piecemeal to inspect and change other readers as others have time/effort/motivation. Or if you're feeling bold you can tackle them all at once -- reviewing won't be much harder since changing each reader will be nicely isolated to its own folder in the source tree.

paulroujansky, 2023-09-07:
> Based on your activity, I'm guessing you're working with (just) EDF for now, is that right?

Haha indeed, well spotted! 😉 But I'd be happy to extend the changes to other formats if they apply of course!

> A PR for this would be welcome

Great! I'd be happy to contribute obviously 😃

PR #12720 (2024-07-17): ENH: Use date and datetime objects
1. Use `datetime.date` for `info["subject_info"]["birthday"]` rather than a year/month/day tuple
2. Use `datetime.datetime` for `info["helium_info"]["meas_date"]`
3. Fix in-memory anonymization for EDF by deleting `edf_info["subject_info"]` and `edf_info["meas_date"]`

Closes #11966
Closes #11327
closed 2024-07-17 by hoechenberger