mne-tools/mne-python #11968: mni.chpo.compute_chpi_locs: The data rank as it exceeds the theoretical rank

zengtaiping, 2023-09-08:
### Description of the problem

When I run the function **mni.chpo.compute_chpi_locs**:
**There is an error: Something went wrong in the data-driven estimation of the data rank as it exceeds the theoretical rank from the info (303 > 69).**

But there is no parameter to set rank.

```
Computing 4385 HPI location guesses (1 cm grid in a 10.7 cm sphere)
/tmp/ipykernel_56666/2454704541.py:12: RuntimeWarning: Something went wrong in the data-driven estimation of the data rank as it exceeds the theoretical rank from the info (303 > 69). Consider setting rank to "auto" or setting it explicitly as an integer.
  chpi_locs = mne.chpi.compute_chpi_locs(raw.info, chpi_amplitudes, verbose=True)
HPIFIT: 5 coils digitized in order 1 2 3 5 4
HPI consistency of isotrak and hpifit is OK.
```
Then when I continue to run **mni.chpo.compute_head_pos function**:
**I always got 0/5 good HPI fits**.
For the detail:
```
/tmp/ipykernel_56666/2454704541.py:16: RuntimeWarning:     t=48.000:    0/5 good HPI fits, cannot determine the transformation (0.39, 0.27, 0.51, 0.26, 0.29 GOF)!
  head_pos = mne.chpi.compute_head_pos(raw.info, chpi_locs)
/tmp/ipykernel_56666/2454704541.py:16: RuntimeWarning:     t=48.010:    0/5 good HPI fits, cannot determine the transformation (0.36, 0.25, 0.43, 0.28, 0.28 GOF)!
  head_pos = mne.chpi.compute_head_pos(raw.info, chpi_locs)
/tmp/ipykernel_56666/2454704541.py:16: RuntimeWarning:     t=48.020:    0/5 good HPI fits, cannot determine the transformation (0.39, 0.26, 0.40, 0.34, 0.31 GOF)!
  head_pos = mne.chpi.compute_head_pos(raw.info, chpi_locs)
/tmp/ipykernel_56666/2454704541.py:16: RuntimeWarning:     t=48.030:    0/5 good HPI fits, cannot determine the transformation (0.32, 0.28, 0.34, 0.26, 0.38 GOF)!
```

### Steps to reproduce

```Python
# extract the HPI coil amplitudes as a function of time
print("start mni.chpo.compute_chpi_amplitudes function")
chpi_amplitudes = mne.chpi.compute_chpi_amplitudes(raw, verbose=True)

# Compute time-varying HPI coil locations from these:
print("start mni.chpo.compute_chpi_locs function")
chpi_locs = mne.chpi.compute_chpi_locs(raw.info, chpi_amplitudes, verbose=True)

# Compute head positions from the coil locations:
print("start mni.chpo.compute_head_pos function")
head_pos = mne.chpi.compute_head_pos(raw.info, chpi_locs)
# Visualise head positions
head_pos_fig = mne.viz.plot_head_positions(
    head_pos,
    mode="traces",
    destination=raw.info["dev_head_t"],
    info=raw.info,
    show=False,
)
```

### Expected results

I could visualize a continuous head position.

### Actual results

There is no data to visualize when I run mne.viz.plot_head_positions.

IndexError                                Traceback (most recent call last)
Cell In[6], line 18
     16 head_pos = mne.chpi.compute_head_pos(raw.info, chpi_locs)
     17 # Visualise head positions
---> 18 head_pos_fig = mne.viz.plot_head_positions(
     19     head_pos,
     20     mode="traces",
     21     destination=raw.info["dev_head_t"],
     22     info=raw.info,
     23     show=False,
     24 )
     25 head_pos_fig.savefig(megDir + "/01_head_motion/head_pos.tif")
     26 # save files and record

File ~/mne-python/1.5.1_0/lib/python3.11/site-packages/mne/viz/_3d.py:247, in plot_head_positions(pos, mode, cmap, direction, show, destination, info, color, axes)
    245 for ii, (quat, coord) in enumerate(zip(use_quats.T, use_trans.T)):
    246     axes[ii, 0].plot(t, coord, color, lw=1.0, zorder=3)
--> 247     axes[ii, 0].set(ylabel=labels[0][ii], xlim=t[[0, -1]])
    248     axes[ii, 1].plot(t, quat, color, lw=1.0, zorder=3)
    249     axes[ii, 1].set(ylabel=labels[1][ii], xlim=t[[0, -1]])

### Additional information

Platform             Linux-5.15.0-82-generic-x86_64-with-glibc2.31
Python               3.11.5 | packaged by conda-forge | (main, Aug 27 2023, 03:34:09) [GCC 12.3.0]
Executable           /home/brain/mne-python/1.5.1_0/bin/python
CPU                  x86_64 (48 cores)
Memory               125.4 GB

Core
├☑ mne               1.5.1
├☑ numpy             1.24.4 (OpenBLAS 0.3.24 with 48 threads)
├☑ scipy             1.11.2
├☑ matplotlib        3.7.2 (backend=module://matplotlib_inline.backend_inline)
├☑ pooch             1.7.0
└☑ jinja2            3.1.2

Numerical (optional)
├☑ sklearn           1.3.0
├☑ numba             0.57.1
├☑ nibabel           5.1.0
├☑ nilearn           0.10.1
├☑ dipy              1.7.0
├☑ openmeeg          2.5.6
├☑ pandas            2.1.0
└☐ unavailable       cupy

Visualization (optional)
...
├☑ mne-features      0.3
├☑ mne-connectivity  0.5.0
├☑ mne-icalabel      0.4
└☑ mne-bids-pipeline 1.4.0

larsoner, 2023-09-08:
It looks like you are probably trying to compute cHPI locations on data that have already been maxfiltered which is unlikely to work.

We shouldn't close this issue, though, because we should probably detect this and at least emit an informative warning. There are cases where it could in theory work (I think) but most of the time it's probably a mistake. So maybe we should actually have `on_maxwell_filtered=...` at some point, but I think a warning to start is probably good enough.

zengtaiping, 2023-09-08:
Thanks for your quick response. 

I tried without being max-filtered.
It recommended me to do this.

`RuntimeWarning: This file contains raw Internal Active Shielding data. It may be distorted. Elekta recommends it be run through MaxFilter to produce reliable results. Consider closing the file and running MaxFilter on the data.
`
This is the data I used: https://drive.google.com/file/d/1p6wM_nEygKitwgpJyKkqll_SMozxzEFR/view?usp=sharing

larsoner, 2023-09-08:
That's true if you plan on analyzing the raw data. But for computing cHPI locations you should indeed process the raw data itself. You can load it with `read_raw_fif(..., allow_maxshield='yes')` to suppress that warning

zengtaiping, 2023-09-08:
Yes, I actually run the code:
`raw = mne.io.read_raw_fif(megDir + "cut1s_run1_raw.fif", preload=True, allow_maxshield=True)
`
However, but still, there is nothing to visualize:
```
IndexError                                Traceback (most recent call last)
Cell In[14], line 18
     16 head_pos = mne.chpi.compute_head_pos(raw.info, chpi_locs)
     17 # Visualise head positions
---> 18 head_pos_fig = mne.viz.plot_head_positions(
     19     head_pos,
     20     mode="traces",
     21     destination=raw.info["dev_head_t"],
     22     info=raw.info,
     23     show=False,
     24 )
     25 head_pos_fig.savefig(megDir + "/01_head_motion/head_pos.tif")
     26 # save files and record

File ~/mne-python/1.5.1_0/lib/python3.11/site-packages/mne/viz/_3d.py:247, in plot_head_positions(pos, mode, cmap, direction, show, destination, info, color, axes)
    245 for ii, (quat, coord) in enumerate(zip(use_quats.T, use_trans.T)):
    246     axes[ii, 0].plot(t, coord, color, lw=1.0, zorder=3)
--> 247     axes[ii, 0].set(ylabel=labels[0][ii], xlim=t[[0, -1]])
    248     axes[ii, 1].plot(t, quat, color, lw=1.0, zorder=3)
    249     axes[ii, 1].set(ylabel=labels[1][ii], xlim=t[[0, -1]])

IndexError: index 0 is out of bounds for axis 0 with size 0
```

jhouck, 2023-09-08:
I think that's what you would get if the cHPI signal was not present in the data. What do you see in `head_pos` and `chpi_locs`?

zengtaiping, 2023-09-09:
[image]

This is the output.
You mean the data lost the cHPI signal. How could I add to the data?

zengtaiping, 2023-09-12:
Thanks very much for the explanation about this problem. I will come back to you later.

larsoner, 2023-09-18:
I'm going to reopen this and re-title it, see https://github.com/mne-tools/mne-python/issues/11968#issuecomment-1711046430

zengtaiping, 2023-09-19:
I am sorry I closed it. Feel free to reopen it.

PR #12038 (2023-10-01): MAINT: Warn when fitting cHPI amplitudes on Maxwell filtered data
Closes #11968

Not convinced we need a `devel.rst` for this since it's just a new warning for people
closed 2023-10-01 by larsoner