mne-tools/mne-python #11973: MAINT: Add steering committee members to CODEOWNERS

larsoner, 2023-09-08:
Hey @mne-tools/mne-python-steering-committee, could each of you please take a moment to open a PR to add yourself to [CODEOWNERS](https://github.com/mne-tools/mne-python/blob/main/.github/CODEOWNERS) in this repo? We would like to start using this as a way to triage issues, PR reviews, and forum posts to people with relevant domain expertise.

Feel free to use as fine-grained or coarse file patterns as you want following the [gitignore syntax](https://git-scm.com/docs/gitignore#_pattern_format) (see also [cheatsheet](https://gist.github.com/jstnlvns/ebaa046fae16543cc9efc7f24bcd0e31)). To make an example, I've opened a PR on behalf of @rob-luke in https://github.com/mne-tools/mne-python/pull/11972. (And @sappelhoff if you're happy with your current set of files, no need to update anything!)

Progress:

- [ ] agramfort
- [ ] wmvanvliet
- [x] rob-luke
- [x] drammock
- [ ] dengemann
- [ ] hoechenberger
- [ ] larsoner
- [ ] adam2392
- [ ] cbrnr
- [ ] bloyl
- [ ] mmagnuski
- [x] sappelhoff
- [ ] alexrockhill
- [ ] jasmainak
- [ ] GuillaumeFavelier
- [x] britta-wstnr
- [ ] mscheltienne

larsoner, 2023-10-02:
Ping @dengemann @mmagnuski @jasmainak do you want to be added somewhere?

jasmainak, 2023-10-03:
please add me here:

https://github.com/mne-tools/mne-python/blob/8051f6ddf68a797ec0c9e1801aea3c73808bba03/.github/CODEOWNERS#L84

I'll make a PR to add myself if something else crops up in the future

drammock, 2023-10-03:
> please add me here:
> 
> https://github.com/mne-tools/mne-python/blob/8051f6ddf68a797ec0c9e1801aea3c73808bba03/.github/CODEOWNERS#L84
> 
> I'll make a PR to add myself if something else crops up in the future

Please do this PR yourself. Can be your first contribution to this week's sprint :)

drammock, 2023-10-03:
https://docs.github.com/en/repositories/managing-your-repositorys-settings-and-features/customizing-your-repository/about-code-owners#about-code-owners

> Code owners are automatically requested for review when someone opens a pull request that modifies code that they own.

mmagnuski, 2023-10-03:
Sorry for missing this, I'll take a look and add myself here and there (separate PR then).

larsoner, 2023-10-06:
Still some left to add, reopening

jasmainak, 2023-10-06:
okay I went ahead and added myself to codeowners [here](https://github.com/mne-tools/mne-python/pull/12089/files)

practically speaking though, I watch all the discussions on mne-python and pitch in when necessary ... everything goes through a filter in the mailbox and is checked periodically. I don't think it's easy to create separate filters for codeowner-triggered notifications vs rest of github.

hoechenberger, 2023-10-06:
@jasmainak be aware that whenever a PR touching one of the parts of the project that you signed up as a code owner for, a review request will be triggered, as @cbrnr mentioned above
This might totally spam your GH Notification center

sappelhoff, 2023-10-24:
cc @mmagnuski @GuillaumeFavelier @bloyl friendly ping --> you're still missing from the file :-)

larsoner, 2024-01-10:
@mmagnuski happy new year! Do you have a minute to add yourself to CODEOWNERS?

larsoner, 2024-07-15:
Closing since this is complete enough I think

PR #12061 (2023-10-03): Revise cbrnr code ownership
Fixes #11973.
closed 2024-07-15 by larsoner