mne-tools/mne-python #11977: Create a dipole fitting GUI that is better than XFit

wmvanvliet, 2023-09-09:
### Describe the new feature or enhancement

Starting an issue to discuss our dream dipole fitting interface. MEGIN's Xfit program has reigned supreme for what seems like forever, but that doesn't mean it's perfect.

Use cases for Xfit (feel free to add your own):

 - [x] View magnetic field patterns for evokeds. Scroll through time. (#11942)
   - [x] We can also do EEG fields.
   - [x] Our scrolling is smoother.
   - [x] Rotating camera is more intuitive than Xfit's 9 panel view. 
 - [x] View head-in-helmet alignment (`plot_alignment` and `Brain.add_head`)
   - [x] We can also plot the brain and reconstructed head surface (verifying MRI alignment) 
 - [x] View sensor-level evokeds in topo form (`plot_evoked_topo`)
   - [x] We can plot multiple evokeds (e.g. multiple conditions) in the same figure
 - [ ] Quickly adjust `ylim` for sensor-level evokeds view (#11892)
 - [ ] Quickly toggle mag/grad/eeg for sensor-level evokeds view
 
- [ ] Fit dipole with the click of a button
- [ ] See dipole projected on the helmet as an arrow
- [ ] See dipole timecourse
  - [ ] See dipole timecourse for multiple evokeds (e.g. multiple conditions) 
- [ ] Select subset of sensors and fit dipole with only those sensors
  - [ ] Use lasso select instead of rectangular box 
- [ ] See magnetic field patterns produced by selected subset of sensors only
- [ ] Fit more than one dipole
- [ ] When computing dipole timecourses, select between single-dipole or multi-dipole modeling
- [ ] See percentage of signal variance explained by fitted dipoles
- [ ] Overlay signal explained by fitted dipoles on top of the sensor-level signals
  - [ ] Do this for multiple evokeds (e.g. multiple conditions) 
- [ ] Turn dipoles on and off
- [x] Save fitted dipoles to file
- [ ] Select whether fitted dipoles are allowed to move in time or not

 - [ ] Select precomputed BEM model
 - [ ] Create sphere BEM model with the click of a button
 - [ ] Select noise-cov or use "standard" one

### Describe your proposed implementation

Use UI event system to tie together existing plotting functions. Enhance existing plotting functions to allow for more interactivity. Finally, create complete GUI with all these pieces.

### Describe possible alternatives

Stick fit Xfit

larsoner, 2023-09-25:
We should consider also how many of these we can fix at the same time (maybe zero but hopefully more than that):

- https://github.com/mne-tools/mne-python/issues/2995
- https://github.com/mne-tools/mne-python/issues/4655
- https://github.com/mne-tools/mne-python/issues/11376

larsoner, 2024-07-15:
@wmvanvliet clearing the milestone on this one but feel free to put it to 1.9 if you think you'll have time in the next release cycle