mne-tools/mne-python #11986: Floating point precision is undefined

mercedesbarrachina04, 2023-09-11:
### Description of the problem

I am trying to upload a Brain Vision file and I am getting the following error: "Floating point precision is undefined". Is this an error in the library?

### Steps to reproduce

```Python
Just using the read_header(fid)and getting the message above
```

### Expected results

Completely read the file

### Actual results

Error message with the text "Floating point precision is undefined"

### Additional information

Platform:         Windows-10-10.0.22621-SP0
Python:           3.9.12 (main, Apr  4 2022, 05:22:27) [MSC v.1916 64 bit (AMD64)]
Executable:       C:\ProgramData\Anaconda3\python.exe
CPU:              Intel64 Family 6 Model 140 Stepping 1, GenuineIntel: 8 cores
Memory:           15.7 GB

mne:              1.2.0
numpy:            1.21.1 {OpenBLAS 0.3.17 with 8 threads}
scipy:            1.10.1
matplotlib:       3.5.1 {backend=module://matplotlib_inline.backend_inline}

sklearn:          1.0.2
numba:            0.57.1
nibabel:          Not found
nilearn:          Not found
dipy:             Not found
openmeeg:         Not found
cupy:             Not found
pandas:           1.4.2
pyvista:          Not found
pyvistaqt:        Not found
ipyvtklink:       Not found
vtk:              Not found
qtpy:             2.0.1 {PyQt5=5.9.7}
ipympl:           Not found
pyqtgraph:        Not found
pooch:            v1.6.0

mne_bids:         Not found
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   Not found
mne_connectivity: Not found
mne_icalabel:     Not found

cbrnr, 2023-09-12:
> Just using the read_header(fid)and getting the message above

There is no `read_header()` function, so can you please share the code that leads to the error (i.e. a minimal example)?

mercedesbarrachina04, 2023-09-12:
> > Just using the read_header(fid)and getting the message above
> 
> There is no `read_header()` function, so can you please share the code that leads to the error (i.e. a minimal example)?

Yes, it is the function def _read_header(fid): you can find the link of the function here: https://github.com/mne-tools/mne-python/blob/main/mne/io/egi/egi.py

cbrnr, 2023-09-12:
This is a private function, you should not use it. Again, can you share your code so that we can reproduce the issue? It looks like you are not using the correct function, because `mne.io.read_raw_egi()` cannot be used to import BrainVision files. Instead, you need to use `mne.io.read_raw_brainvision()`. Alternatively, you can use `mne.io.read_raw()` for any supported file type.

closed 2023-09-12 by mercedesbarrachina04