mne-tools/mne-python #11998: Bad split size with raw.save

mscheltienne, 2023-09-14:
### Description of the problem

In today's weird bug, I have a MEGIN recording `test_sleep_raw.fif` splitted in 2 files (2 GB + 1.3 GB). The I/O roundtrip through MNE-Python to crop the 40 first second before running MEGIN's MaxFilter yields 2 files (so far so good), of size 2.1 GB + 1.2 GB, despite the default `split_size='2GB'` argument.

```
from mne.io import read_raw_fif

raw = read_raw_fif("test_sleep_raw.fif", preload=True)
raw.del_proj(len(raw.info["projs"]) - 1)
raw.crop(40, None)
raw.save("test_sleep_cropped_raw.fif", drop_small_buffer=True)
```

And MaxFilter doesn't like that the 2 145 MB size of the first file :cry: 

| ERROR: Too large output file (2145 MB)!

Tested on MNE-main.

### Steps to reproduce

```Python
-
```

### Expected results

-

### Actual results

-

### Additional information

-

larsoner, 2023-09-14:
@mscheltienne can you share the file privately? If not then I suggest you look here since I think this is when we decide to move on to the next file:

https://github.com/mne-tools/mne-python/blob/e6afe5fb64c46df50256635b74cf583d5cc92727/mne/io/base.py#L2793-L2795

larsoner, 2023-09-14:
... one possibility is that `_NEXT_FILE_BUFFER` is not big enough for your file for some reason. If so, we could improve the logic of that by first figuring out what's so big that's being written and account for it. My guess is that it could be the tree/dir tags written at the end of the file, in which case we could try to estimate how much space this will take rather than relying on `_NEXT_FILE_BUFFER` for example by writing it to a dummy BytesIO before the loop begins, then updating the value knowing that every time we write another data buffer we'll need X number of additional bytes when we write our tags at the end.

mscheltienne, 2023-09-15:
Thanks for the tips! I'll have a look early next week; and if I don't figure it out, I'll share the file.