mne-tools/mne-python #12022: Add mne.Forward.save and enable saving large Forward solution > 2 GB

mscheltienne, 2023-09-28:
### Description of the problem

Surprisingly, there is no method `Forward.save` to shortcut `mne.write_forward_solution`.
Also, I got a fairly large forward solution (don't ask..), and it does not fit in a single FIFF file, limited to 2 GB. The current writer is not able to write this forward solution. 

### Steps to reproduce

```Python
-
```

### Expected results

-

### Actual results

-

### Additional information

-

larsoner, 2023-09-28:
We could work out splitting but it would be a pain. Would HDF5 be okay like we added for source spaces in https://github.com/mne-tools/mne-python/issues/4831 ?

mscheltienne, 2023-09-28:
HDF5 sounds good!

PR #12036 (2023-10-01): ENH: Add Forward.save and hdf5 support
Closes #12022
Closes #12031 

No need for latest.inc update for #12031 I think because it's more or less an improvement to the error that gets raised

@mscheltienne feel free to mark for merge-when-green if you're happy. Feel free to test it with your gigantic forward solution but in principle if it fails it's some HDF5 bug.
closed 2023-10-01 by larsoner