mne-tools/mne-python #12024: ENH: Switch to constrained layout in figure creation

larsoner, 2023-09-28:
I've mentioned this a few times in passing (e.g., [1](https://github.com/mne-tools/mne-python/issues/7751#issuecomment-660478066), [2](https://github.com/mne-tools/mne-python/pull/11867#issuecomment-1668358648)) but thought it deserves its own issue. I've been using `plt.subplots(..., constrained_layout=True)` for a few years in many use cases and I can't think of a time it produced a bad (or even sub-optimal) result. By comparison, the default `tight_layout` engine and workflow almost always needs some manual adjustment to use space optimally.

I propose we try switching our `plt.subplots` and related matplotlib-figure-creation calls to use the constrained layout engine wherever possible, build the doc, and see if it does indeed improve things. In particular I expect it:

1. Will produce better results more consistently in (nearly?) all cases compared to current code that calls `subplots_adjust` and `tight_layout`
2. Is easy enough for people to get old behavior back by switching the layout engine on their own with `fig.set_layout_engine('tight')` and/or `fig.tight_layout` etc.
3. Will allow us to simplify our code because we won't need to tweak plot margins etc. nearly as much (or at all) anymore, just make minimal decisions (e.g., how much space do I want to reserve for the colorbar?)
4. By removing all our internal uses of `subplots_adjust` / `tight_layout`, if a user passes in their own `axes` we'll leave them alone rather than tweak the figure properties, which seems like cleaner/nicer behavior. (If a user passes `axes` it should be up to them to deal with spacing issues however they wish without us messing with the spacing.)

So I expect an overall meaningful aesthetic and maintenance win by going this route. Since we consider aesthetic changes non-breaking I don't think we need a deprecation cycle for this. If we anticipate user backlash, we could add a `mne.viz.use/get/set_figure_layout_engine`, but I'd assume YAGNI and just have users do `fig.set_layout_engine('tight')` when they need it.

Bringing this up was inspired by @mscheltienne going this direction a bit already in #12020.

larsoner, 2023-09-28:
@mscheltienne maybe one for you for the sprint if this ends up being uncontroversial and you're motivated :)

hoechenberger, 2023-09-28:
Let's do this

PR #12050 (2023-10-07): Use constrained layout in matplotlib visualization
Closes #12024

Bump the minimum requirement for `matplotlib` to 3.5.0, which supports the recommended `layout="constrained"` argument and which will be out for 2 years on the 16th of November.

- [x] Get CIs green
- [x] eradicate `subplots_adjust`
- [x] Run [circle full]
- [x] Visual inspection
closed 2023-10-07 by drammock