mne-tools/mne-python #12047: MAINT: Add pyi files for modules

larsoner, 2023-10-02:
> Just to emphasize one point from above: unless lazy loading happens in `.pyi` files, type checkers won't know about those functions. See:
>
> https://scientific-python.org/specs/spec-0001/#type-checkers

_Originally posted in https://github.com/mne-tools/mne-python/issues/11780#issuecomment-1742344182_

I wanted to separate this from #11780, I think @drammock might give it a shot?

hoechenberger, 2023-10-02:
Just to add, lazy loading currently also breaks the VS Code Python debugger (debugpy). It just throws an `AttributeError` and exits.

larsoner, 2023-10-02:
We'll have to check to see if the `.pyi` file fixes this, too. If not we should raise an issue on the `lazy_loader` tracker after checking to see if there already is one. Seems like a caveat that should definitely be listed if it's indeed a drawback with no workaround (e.g., waiting on a vscode fix that might never come)

hoechenberger, 2023-10-02:
I can take a look at this later today or tomorrow.

I also wanted to ask which approach we're planning to take here. We could add `.pyi` files inside MNE-Python, or we could add stubs to `typeshed` – and also slowly start adding type annotations there. I've been wanting to do that for a long time and started (and stopped) many times before.

hoechenberger, 2023-10-02:
Also see this approach for resolving the problem **without** adding `.pyi` files:
https://github.com/microsoft/pylance-release/issues/474#issuecomment-706719841

larsoner, 2023-10-02:
To me it makes sense to have `.pyi` to be part of this repo rather than live in `typeshed`

drammock, 2023-10-02:
> I can take a look at this later today or tomorrow.

great, I'll remove myself and assign you 🙏🏻

PR #12072 (2023-10-05): handle lazy loading through .pyi type stubs
closes #12059 (supersedes)
fixes #12047
closed 2023-10-05 by larsoner