mne-tools/mne-python #12048: Lazy loading makes working with MNE-Python scripts in VS Code very awkward

hoechenberger, 2023-10-02:
Hello, I'm using VS Code with Pylance, and since lazy loading was introduced, using MNE is no fun at all anymore. It appears that Pylance doesn't know about any of those "lazy" imports, providing a subpar development experience:

[image]

Pylance is the default Python type & syntax checker in VS Code, which also provides completion suggestions etc.

A workaround could be to distribute (type) stubs via `typeshed`, which is [what TensorFlow does](https://github.com/microsoft/pylance-release/issues/3174) to deal with this exact same problem.

hoechenberger, 2023-10-02:
Potential alternative approach:
https://github.com/microsoft/pylance-release/issues/474#issuecomment-706719841

larsoner, 2023-10-02:
Closing as a dup of #12047 since I am pretty sure it will solve this problem (but we should definitely reopen this if it doesn't)

closed 2023-10-02 by larsoner

hoechenberger, 2023-10-02:
Thanks @larsoner, I didn't realize you'd just opened this issue :)