mne-tools/mne-python #12091: mne.time_frequency.tfr_array_morlet contiguous error

RainyMeadows, 2023-10-08:
### Description of the problem

Running on a multi-epoch input returns `Python Error: ValueError: ndarray is not contiguous`

### Steps to reproduce

```Python
import mne
import numpy as np
# data is 20 Epochs * 32 Channels * 12000 Times
freqs = np.logspace(np.log10(2),np.log10(100),46)
Power_Event_1 = mne.time_frequency.tfr_array_morlet(data(1,:,:), 1000, freqs, n_cycles=6, output='power')
Power_Event_2 = mne.time_frequency.tfr_array_morlet(data(2,:,:), 1000, freqs, n_cycles=6, output='power')
Power_At_Once = mne.time_frequency.tfr_array_morlet(data(1:2,:,:), 1000, freqs, n_cycles=6, output='power')
```

### Expected results

All three outputs ought to be okay.

### Actual results

Power_Event_1, and Power_Event_2 are okay. Error for Power_At_Once 

### Additional information

Platform:         Windows-10-10.0.22621-SP0
Python:           3.10.11 (tags/v3.10.11:7d4cc5a, Apr  5 2023, 00:38:17) [MSC v.1929 64 bit (AMD64)]
Executable:       C:\Users\DELL\AppData\Local\Programs\Python\Python310\python.exe
CPU:              Intel64 Family 6 Model 151 Stepping 2, GenuineIntel: 20 cores
Memory:           31.7 GB

mne:              1.3.1
numpy:            1.24.3 {unknown linalg bindings (threadpoolctl module not found: No module named 'threadpoolctl')}
scipy:            1.10.1
matplotlib:       3.7.1 {backend=TkAgg}

sklearn:          Not found
numba:            Not found
nibabel:          Not found
nilearn:          Not found
dipy:             Not found
openmeeg:         Not found
cupy:             Not found
pandas:           2.0.3
pyvista:          Not found
pyvistaqt:        Not found
ipyvtklink:       Not found
vtk:              Not found
qtpy:             Not found
ipympl:           Not found
pyqtgraph:        Not found
pooch:            v1.7.0

mne_bids:         Not found
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   Not found
mne_connectivity: Not found
mne_icalabel:     Not found

larsoner, 2023-10-09:
For `Steps to reproduce` I don't see any definition of `data`, can you add one? For example if you use `data = np.random.RandomState(0).randn(...)` is there a problem? Or is there something different about about how your `data` are constructed?

drammock, 2023-10-09:
in addition to @larsoner's questions, I will point out that

1. your example code doesn't look right, e.g., `data(:,8:9,:)` implies that data is a *function* not an *array* (shouldn't it be `data[:,8:9,:]`?)
2. your variable names are misleading, `Power_Both_Event_All_Chan` and `Power_Event_All_Two_Chan` imply that they will contain two elements along a given axis, but in fact they do not (Python slice indexes exclude the second element, so `1:2` includes element 1 but not element 2, and likewise for `8:9`).