mne-tools/mne-python #12092: MNE Epochs notch_filter feature request

hyphenzhao, 2023-10-09:
### Describe the new feature or enhancement

There is notch_filter function in Raw object. However the Epochs object does not have this function. I have seen discussion in forum and it seems a missing instead of not-implemented on purpose. I don't see this as a complex feature and how did we miss that? A little bit more context about requiring this feature:
I have multiple datasets with various formats. And I will have to epochs them first to align the format before applying notch filter. Since I have to clean the events/triggers manually, if I apply resample before epochs, the events would be very tricky to align with the resampled raw data. As a result, the notch filter will be slow with orignal data. Therefore, the notch filter of epoched data seemed to be important and essential for saving calculation time in my case.  

### Describe your proposed implementation

As epochs are discontinuous data, I would propose to treat each epoch as a single fragment and apply notch filter. Perhaps the function could require the minimum length of every epoch depending on the desired notch filter range. This could prevent potential errors. 

### Describe possible alternatives

Alternatively, I would appreciate it if anyone could kindly explain the issues behind implenmenting this.

### Additional context

The additional context was specified previously.

agramfort, 2023-10-10:
do you know that you have an events parameter in raw.resample
https://mne.tools/stable/generated/mne.io.Raw.html#mne.io.Raw.resample

so that you can readjust events for the new sampling rate

> Message ID: ***@***.***>
>

larsoner, 2023-10-10:
There are going to be issues with filter length but these also occur for regular filtering, so I guess we could add `.notch_filter` to Epochs using a Mixin. We probably shouldn't allow the `spectrum_fit` method there, though -- just FIR/IIR.

hyphenzhao, 2023-10-11:
> do you know that you have an events parameter in raw.resample https://mne.tools/stable/generated/mne.io.Raw.html#mne.io.Raw.resample so that you can readjust events for the new sampling rate
> […](#)
> Message ID: ***@***.***>

Thanks! I didn't know that before. That really resolves my current issue. Nevertheless, it would be neat if I could do notch filter after epochs as in that case I could ask others to epoch their data to same format for my analysis script to run in the future. Anyway, it shouldn't be too complicated to perform notch filter on Raw object and then epoch with various file format. Thanks. 

> There are going to be issues with filter length but these also occur for regular filtering, so I guess we could add `.notch_filter` to Epochs using a Mixin. We probably shouldn't allow the `spectrum_fit` method there, though -- just FIR/IIR.

If that is the case, then a Mixin notch_filter could be an option maybe? An option would be nice and cautious could be made in the API reference. May I ask if you have any plan to add this feature? Either way, I could close the issue with your response on this question. Thanks!
Regards,
Hyphen.

larsoner, 2023-10-11:
> That really resolves my current issue. Nevertheless, it would be neat if I could do notch filter after epochs as in that case I could ask others to epoch their data to same format for my analysis script to run in the future.

For now let's try to get away without it. Results will be better if you filter `raw` first, so let's try to stick with that. But if other people (or you!) do end up needing epochs-level notch filtering later let's reopen the issue and implement it.

closed 2023-10-11 by larsoner