mne-tools/mne-python #12094: Emmit an error if you try to use cHPI functions on data with cHPI disabled

mscheltienne, 2023-10-10:
### Describe the new feature or enhancement

Hello, we notice today that you can run without any error cHPI functions on data which did not have cHPI enabled.
It does return empty arrays, or arrays filled with nans, but an error message would be less confusing. 

Do we have a definitive way to find out if cHPI was on or off during a recording in an Info?

### Describe your proposed implementation

-

### Describe possible alternatives

-

mscheltienne, 2023-10-10:
Out of curiosity, do you know where we would detect that in new files?

larsoner, 2023-10-10:
For MEGIN data there is a cHPI on status channel. If you search "_on" in cHPI.py I think you'll find it. If it's present and always off (or always < 3 active I guess) we could warn/raise early (again with option)

mscheltienne, 2023-10-10:
It does not seem very informative.
Comparing a file with and without cHPI, in both case `get_chpi_info` returns `hpi_on` (suppose to be "The values coding for the “on” state of each individual cHPI coil.") as `array([ 256,  512, 1024, 2048, 4096], dtype=int32)`. 
The SYS201 channel which should have some information on the interval of the cHPI measurements contains a flat `7936` value after cHPI is enabled. 
If I look at `get_active_chpi`, this check is suppose to find if we have cHPI on or off: https://github.com/mne-tools/mne-python/blob/77328c09b8f709bee2c702d66cc8935fcf8c83d5/mne/chpi.py#L1618-L1621

But in both cases, the `if` statement is False and that early return is by-passed.

Now I'm also wondering what would happen if I plugged the HPI connector into the HPI connector 2 instead of 1.. Anyway, for sure catching the emptiness..

larsoner, 2023-10-10:
> But in both cases, the if statement is False and that early return is by-passed.

There should be a later check where we apply the `array` bitmask to the SYS201 channel, so:
```
>>> (7936 & np.array([256,  512, 1024, 2048, 4096])).astype(bool).sum()
5
```
If your SYS201 channel has this value for all time, it means your cHPI was on for the whole file. If you compare this to a file where cHPI was turned on only briefly:
```
>>> raw = mne.io.read_raw_fif(mne.datasets.brainstorm.bst_phantom_elekta.data_path() / 'kojak_all_200nAm_pp_no_chpi_no_ms_raw.fif')
>>> mne.chpi.get_active_chpi(raw)
array([0, 0, 0, ..., 0, 0, 0])
>>> mne.chpi.get_active_chpi(raw).mean()
0.028565217391304347
```
because they only turned it on for a tiny part in the beginning as you can see here:

https://mne.tools/dev/auto_tutorials/inverse/80_brainstorm_phantom_elekta.html#sphx-glr-auto-tutorials-inverse-80-brainstorm-phantom-elekta-py

> I'm also wondering what would happen if I plugged the HPI connector into the HPI connector 2 instead of 1..

Probably nothing in the file would tell you this other than the amplitudes and cHPI SNR being terrible :(

larsoner, 2023-10-10:
... but also the step where you estimate `dev_head_t` at the beginning of the run during acquisition itself should have failed if you used the wrong port.