mne-tools/mne-python #12106: 运行sleep60代码但报错

Summer-77, 2023-10-12:
### Description of the problem

Traceback (most recent call last):
  File "D:\pythonProject\Brain\model1\file1.py", line 43, in <module>
    import mne
  File "D:\Ananconda\envs\py3.11\lib\site-packages\mne\__init__.py", line 44, in <module>
    from .io.pick import (
  File "D:\Ananconda\envs\py3.11\lib\site-packages\mne\io\__init__.py", line 24, in <module>
    from .base import BaseRaw
  File "D:\Ananconda\envs\py3.11\lib\site-packages\mne\io\base.py", line 32, in <module>
    from ..channels.channels import (
  File "D:\Ananconda\envs\py3.11\lib\site-packages\mne\channels\__init__.py", line 15, in <module>
    from .montage import (
  File "D:\Ananconda\envs\py3.11\lib\site-packages\mne\channels\montage.py", line 24, in <module>
    from ..viz import plot_montage
  File "D:\Ananconda\envs\py3.11\lib\site-packages\mne\viz\__init__.py", line 64, in <module>
    from .epochs import plot_drop_log, plot_epochs, plot_epochs_psd, plot_epochs_image
  File "D:\Ananconda\envs\py3.11\lib\site-packages\mne\viz\epochs.py", line 20, in <module>
    from .raw import _setup_channel_selections
  File "D:\Ananconda\envs\py3.11\lib\site-packages\mne\viz\raw.py", line 16, in <module>
    from ..time_frequency import Spectrum
  File "D:\Ananconda\envs\py3.11\lib\site-packages\mne\time_frequency\__init__.py", line 32, in <module>
    from ._stockwell import tfr_stockwell, tfr_array_
ImportError: cannot import name 'tfr_array_' from 'mne.time_frequency._stockwell' (D:\Ananconda\envs\py3.11\lib\site-packages\mne\time_frequency\_stockwell.py)

### Steps to reproduce

```Python
不知道怎么解决
```

### Expected results

修改不了

### Actual results

无

### Additional information

windows11，利用pycharm，python=3.11

mscheltienne, 2023-10-12:
Hello, could you open issues in english and could you provide the output of `mne.sys_info()` as well?

Summer-77, 2023-10-12:
> 您好，您能否用英语打开问题，您能否提供输出？`mne.sys_info()`
Excuse me, can you be a little more specific? What is this problem? How to modify

closed 2023-10-12 by Summer-77