mne-tools/mne-python #12107: BUG

Summer-77, 2023-10-12:
Run sleep60 model  has this bug，i could not   solve    
  Traceback (most recent call last):
  File "D:\pythonProject\Brain\model1\file1.py", line 43, in <module>
    import mne
  File "D:\Ananconda\envs\py3.11\lib\site-packages\mne\__init__.py", line 44, in <module>
    from .io.pick import (
  File "D:\Ananconda\envs\py3.11\lib\site-packages\mne\io\__init__.py", line 24, in <module>
    from .base import BaseRaw
  File "D:\Ananconda\envs\py3.11\lib\site-packages\mne\io\base.py", line 32, in <module>
    from ..channels.channels import (
  File "D:\Ananconda\envs\py3.11\lib\site-packages\mne\channels\__init__.py", line 15, in <module>
    from .montage import (
  File "D:\Ananconda\envs\py3.11\lib\site-packages\mne\channels\montage.py", line 24, in <module>
    from ..viz import plot_montage
  File "D:\Ananconda\envs\py3.11\lib\site-packages\mne\viz\__init__.py", line 64, in <module>
    from .epochs import plot_drop_log, plot_epochs, plot_epochs_psd, plot_epochs_image
  File "D:\Ananconda\envs\py3.11\lib\site-packages\mne\viz\epochs.py", line 20, in <module>
    from .raw import _setup_channel_selections
  File "D:\Ananconda\envs\py3.11\lib\site-packages\mne\viz\raw.py", line 16, in <module>
    from ..time_frequency import Spectrum
  File "D:\Ananconda\envs\py3.11\lib\site-packages\mne\time_frequency\__init__.py", line 32, in <module>
    from ._stockwell import tfr_stockwell, tfr_array_
ImportError: cannot import name 'tfr_array_' from 'mne.time_frequency._stockwell' (D:\Ananconda\envs\py3.11\lib\site-packages\mne\time_frequency\_stockwell.py)

mscheltienne, 2023-10-12:
Once again, issues have a template with required information.
If you just dump a traceback without any context, any information from your platform and versions, we can not help you and fix the bug. 
Please follow the template, describe the issue, describe your system configuration (run `mne.sys_info()`) and if possible provide a minimally working example, i.e. a code to copy/paste that we can run and that will exhibit the same traceback.

Mathieu

drammock, 2023-10-12:
@mscheltienne the traceback is occuring on `import mne` so we shouldn't expect `mne.sys_info()` to even be possible here.

@Summer-77 I think something is wrong with your MNE-Python installation. Can you explain how you installed it?  Did you use [our standalone installer](https://mne.tools/stable/install/installers.html), do a [manual install](https://mne.tools/stable/install/manual_install.html), or set up a [development environment](https://mne.tools/stable/install/contributing.html#setting-up-your-local-development-environment)? Or something else?

closed 2024-01-24 by mscheltienne