mne-tools/mne-python #12108: Inferring preprocessing performed on original data from a given MNE object

Genuster, 2023-10-13:
### Describe the new feature or enhancement

Given a saved-by-MNE .fif file I would like to infer what preprocessing procedures were applied to the original recording.
There already exist fields in the Info structure like `highpass`, `lowpass`, `sfreq` that track current state of the data, but there is no way to check if and how they were different before preprocessing.

In addition, there is no way to know what channels, if any, were interpolated, and whether the EEG reference was changed (and also what was the original one). Perhaps, removal of independent components or can also be somehow tracked?

### Describe your proposed implementation

Starting from @larsoner's [comment](https://github.com/mne-tools/mne-python/issues/11705#issuecomment-1561110794) regarding the Info object:
> `info` should really be static and have to do with the acquisition setup itself. Characteristics of the recorded data values (peak to peak, flatness, etc.) shouldn't live there. In this sense it's a bit weird that `info['bads']` exists but we should try to avoid making the problem worse :)

In addition to `bads` I'd mention that `highpass`, `lowpass`, `sfreq` are also dynamic values.

I suggest adding another `Info`-like structure to store all the dynamic, preprocessing-related information, as opposed to static, acquisition-related one.

### Describe possible alternatives

Perhaps, all these preprocessing values can be stored as direct attributes of MNE objects. 
But it is very likely that there are additional dynamic features that I haven't thought about and they'll very fast create a clutter.

larsoner, 2023-10-13:
There are many steps that can be taken in many orders with repeats etc. Provenance is something I'd prefer to build into downstream tools like [mne-bids-pipeline](https://mne.tools/mne-bids-pipeline/1.4/index.html) / https://github.com/mne-tools/mne-bids-pipeline/issues/763 where we would get it very easily from the pipeline structure.

Genuster, 2023-10-13:
I've checked out BEP028, very nice. If a global bidsification is indeed coming, then it makes sense to stick with them. Thanks, @larsoner!

closed 2023-10-13 by Genuster