mne-tools/mne-python #12112: Unit Conversion Helper function for eye-tracking data (pixels to radians)

scott-huberty, 2023-10-16:
### Describe the new feature or enhancement

This Ticket is related to meta-issue #11879

I am at a point in one of my projects where I'd like to convert my eye-tracking position data from pixels-on-screen to a unit of angle. Since 1) this is a fairly common conversion in eye-tracking research; 2) it was highlighted as a TODO in #11879 ; and 3) this would help users convert their position data into an SI unit (radians), I think It would be good to build a helper function for this.

CC @britta-wstnr @larsoner 

### Describe your proposed implementation

Possible API (borrowing from @drammock):

```
mne.preprocessing.eyetracking.convert_units(calibration: Calibration, from: str = 'px', to: str = 'radian')
```
Where `Calibration` is an instance of `mne.preprocessing.eyetracking.Calibration` that contains necessary metadata: `screen_size`, `distance`, `screen_resolution`.

Or we could have a more specific function like:
```
mne.preprocessing.eyetracking.convert_pixels_to_radians(calibration: Calibration)
```

and a similar function for `radian` to `pixel`

### Describe possible alternatives

We could also include `screen_size`, `screen_resolution`, `distance` as parameters to the function, so the user can either pass a Calibration instance with this info, or specify them directly via the function call (this was an idea brought up in the summer and documented in [this comment](https://github.com/mne-tools/mne-python/pull/11719#issuecomment-1584822131))

### Additional context

This was a feature that I proposed during my GSoC but didn't get to by the end of summer.

mscheltienne, 2023-10-19:
```
mne.preprocessing.eyetracking.convert_units(calibration: Calibration, from: str = 'px', to: str = 'radian')
```

with optional `screen_size`, `screen_resolution` and `distance` sounds good to me.

britta-wstnr, 2023-10-19:
Yeah I agree that `distance` and such should probably be parameters. 
One thing to think about then: if a parameter is specified in `Calibration` but also gets passed and numbers don't match, what should be expected behavior? Overwrite by parameters and throw warning? Throw error?

PR #12237 (2024-02-06): Function to convert eyegaze units to radians
closes #12112
closes #12391 
This PR implements a function `mne.preprocessing.eyetracking.convert_units` to help users convert pixel data to radians, which is nice for us because radians are an SI unit, and it may come in handy down the line because converting pixel data to degrees/radians of visual angle is often done prior to calculating eye-tracking related derivatives (saccade amplitude, saccade velocity, etc). 

Perhaps the best way to make sure this is behaving as expected is to plot a heatmap before and after unit conversion. But i'll need to [... cut]
closed 2024-02-06 by larsoner