mne-tools/mne-python #12126: How do I get the "subjects_dir" on my own EEG data?

Sarahsus, 2023-10-24:
### Description of the problem

In the following code:
brain = stc.plot(
    subjects_dir=subjects_dir,
    initial_time=initial_time,
    clim=dict(kind="value", lims=[3, 6, 9]),
    smoothing_steps=7,
)

Now I have some subject EEG data. How do I get the corresponding subject folder,like "subject_dir"?

### Steps to reproduce

```Python
-
```

### Expected results

-

### Actual results

-

### Additional information

-

hoechenberger, 2023-10-24:
Hello @Sarahsus, the subjects dir here refers to the FreeSurfer subjects directory. It has nothing to do with your EEG data per se. 

This question is better suited for our forum, may I kindly refer you there? Thank you!

https://mne.discourse.group/

Richard

closed 2023-10-24 by hoechenberger

Sarahsus, 2023-11-08:
> 日　期：2023年10月24日 17:30:16
> 主　题：Re: [mne-tools/mne-python] How do I get the "subjects_dir" on my own EEG data? (Issue #12126)

thank you

来自钉钉专属商务邮箱------------------------------------------------------------------

https://mne.discourse.group/