mne-tools/mne-python #12130: Info representation in Report is broken

hoechenberger, 2023-10-24:
### Description of the problem

In the Report, the sections of all Info objects are collapsed, but the vertical space is still occupied

### Steps to reproduce

```Python
# %%
import mne

sample_dir = mne.datasets.sample.data_path()
sample_fname = sample_dir / 'MEG' / 'sample' / 'sample_audvis_raw.fif'

raw = mne.io.read_raw_fif(sample_fname)
raw.crop(tmax=60)

report = mne.Report()
report.add_raw(raw, title="Test")
report.save("/tmp/report.html", overwrite=True)
```

### Expected results

- All sections should be uncollapsed by default
- Collapsing a section should free the corresponding vertical space

### Actual results

[image]

### Additional information

Report has its own collapsible sections

For the purpose of quick data / report assessment, all Info sections should be uncollapsed by default

larsoner, 2023-10-24:
> For the purpose of quick data / report assessment, all Info sections should be uncollapsed by default

Agreed this should be the case for `mne.Report`. I think for general object `__repr_html__` outside of `mne.Report` we decided to collapse by default in some issue/PR.

cbrnr, 2023-10-24:
Collapsing in notebooks doesn't make a lot of sense, because e.g. `raw.info` will only show section headers without any information at all.

hoechenberger, 2023-10-26:
JS inspector in Safari on macOS says:

```
[Error] ReferenceError: Can't find variable: $
	Global Code (30_info.html:1323)
```

The respective line is:
```html
<script type="text/javascript">$(function () { $("time.localized").each(function () { var el = $(this); el.text(new Date(el.attr("datetime")).toLocaleString([], {dateStyle: "medium", timeStyle: "long"})); }); } )</script></p></div>
```

larsoner, 2023-10-26:
Not related, making new issue

larsoner, 2023-10-26:
Even after #12145 there is still a bug on Safari on macOS with the collapsing in rendered tutorial HTML, see:

https://github.com/mne-tools/mne-python/pull/12145#issuecomment-1781567722

PR #12145 (2023-10-31): ENH: Collapse only in doc gen
Closes #12130
Closes #12131 
Closes #12149

If we eventually want a way to control this behavior we could add env var, context manager, etc. but not implemented here (YAGNI from #12131). Locally seems to work:

[image]

[image]
closed 2023-10-31 by larsoner