mne-tools/mne-python #12132: DOC: Actively suggest updating MNE to a supported version

larsoner, 2023-10-24:
I often have people come to me locally with bugs, only to find that they are on some old version of MNE-Python. I think it's not generally known to neuroscience people that upgrading is important for bug fixes etc., so it would be good to make it clear what the software support policy is and that upgrading is important.

We should document this clearly as much as possible, but I have two other ideas that will actively "pop up" / ping users more actively:

1. Add a date-based warning *on import*. For example, if we release 1.6 on Nov 1 2023, add to `mne/__init__.py` code to emit a warning on Nov 1 2024 that a new version of MNE is available and that they should update to avoid bugs (or some similar explicit message). We don't absolutely know or guarantee our release cadence, but the longest we have gone between releases in the last few years is probably like ~8 mo, and I think we now shoot for ~6 mo -- so to be safe if can make it year from when we cut a release. This could be controllable with a `MNE_CHECK_VERSION_DATE` env var or something in case people have wacky datetime settings.
2. Query the current release version over the internet when `mne.sys_info` is called with a new kwarg `check_version=True`. We don't want our software to start trying to "phone home" without a user asking in some way (i.e., definitely *not* at import time). Then the question becomes what should the default value of `check_version` be:

   - `False`: least invasive but also least informative for users. In our bug report forms and forum stuff we ask people to call `mne sys_info --check-version` or `mne.sys_info(check_version=True)` or so. But the probability of users seeing an informative warning goes down.
   - `True`: most invasive but also most informative for users.
   - `None`: in between if we make it mean "try to check and if it takes more than 2 sec don't bother" or something. (FWIW, `mne sys_info` on my machine currently takes ~3.7s.)

   Whichever way we go, the default behavior for `None` could be controllable with `MNE_SYS_INFO_CHECK_VERSION` var in case people want the default `None` behavior to mean `False` or `True`.

Happy to hear thoughts on these and other ideas for suggesting people update to supported versions.

mscheltienne, 2023-10-24:
+1, I often have people come to me with bugs already fixed.

drammock, 2023-10-24:
I'd guess that the calendar-based approach would probably do a better job of *preventing* the situation you describe @larsoner, as it seems more likely to get people to *proactively* stay up to date. But it's also more invasive so I'm a bit wary of annoying people --- one of my pet peeves with R is the ~7 lines of output it spits on your console every time you fire it up; to do this well we'd probably need to include not just the warning but also guidance on how to update (even if just a URL) and guidance on how to silence the version check.

if we the sys_info approach, I'm also happy because it's more accurate, and easy to do:

```py
import json
from packaging import version
from urllib import request
with open("https://api.github.com/repos/mne-tools/mne-python/releases/latest", "rb") as f:
    response = json.load(f)
print(version.parse(response["tag_name"]))
```

larsoner, 2023-10-24:
> to do this well we'd probably need to include not just the warning but also guidance on how to update (even if just a URL) and guidance on how to silence the version check.

I think both of these are pretty easily doable. For the first we should point to our updating page URL, and for the second we can add a config/env var.

cbrnr, 2023-10-25:
I'm :-1: on the import warning, because this impacts reproducibility if a perfectly fine script suddenly starts to emit warnings in the future. The chance of annoying people is also pretty high IMO, so controlling that with an env var is not a good solution either.

I like the `mne.sys_info()` approach though, but if we go for `check_version=True` by default, we should really introduce something like a deprecation cycle, otherwise people might run into the same reproducibility issues when they use `mne.sys_info()` in their scripts.

britta-wstnr, 2023-10-25:
I agree with @cbrnr that this impacts reproducibility. Especially for users who do not want to update MNE within a project (e.g. because they cannot easily re-run all analyses with the new version due to their computational resources).

It seems to me as this is mostly useful if a user runs into a bug - so maybe it's really about education i.e. documentation? It seems like an easy first step could be to update our BUG template on GitHub to prominently tell the user to check behavior with the latest version.

drammock, 2023-10-25:
> I think both of these are pretty easily doable.

sure, easily doable. I was just making the point that for folks (like me) who are annoyed by a bunch of extra terminal output, it will be for us at least 3 lines.

> if we go for check_version=True by default, we should really introduce something like a deprecation cycle, otherwise people might run into the same reproducibility issues

I'm not sure this point about the deprecation cycle makes sense; if we start with `check_version=None` then *all* users will see a warning when they run `sys_info` unless/until they update their script to say `check_version=False`. On the other hand, if we start out with `check_version=True` then *only* users on old versions will see a warning, and even then *only* once 1.6 (current dev) becomes "old".

larsoner, 2023-10-25:
> It seems to me as this is mostly useful if a user runs into a bug

That's part of it. But also our code will always have some fraction of hidden silent bugs, so people hit these unknowingly. So the warning will also help people who hit those and don't know it. Someone can think their script is perfectly fine, and while *their code* could be, *our software* could have some bug that they never know is fixed, so they don't bother to update and live with some errant behavior unknowingly.

But it sounds like others think the cost (e.g., having to set an env var to avoid a warning) isn't worth the potential benefits here so I'll drop (1).

Sounds like people are generally in favor of some form of (2) so I'll probably give that one a shot for 1.6.

drammock, 2023-10-25:
API idea:
```
mne.sys_info(...,
             check_version = False,   # never check
             check_version = "info",  # print as part of the sys_info output
             check_version = "warn",  # same as "info" but also log a warning if outdated
             )
```
maybe with `True` being an alias for "info"?  Add'l output might be something like
```
Core
├☑ mne               1.6.0 (outdated, current stable is 1.8.1)
```

hoechenberger, 2023-10-25:
> sure, easily doable. I was just making the point that for folks (like me) who are annoyed by a bunch of extra terminal output, it will be for us at least 3 lines.
>
>
>
>
>
>
> I'm not sure this point about [...]

Yes but let's be honest, we're constantly breaking user code and the short deprecation cycles we currently have are more an alibi to soothe ourselves (the devs) than really helping users actively using our software. The cycles are waaaay too short, for example in a "production" environment it's not unusual to skip one or maybe even two or three versions before updating again, and then your scripts are already non-functional because you missed the deprecation warnings.

hoechenberger, 2023-10-25:
We could fully embrace SemVer and bump the major version every time the API changes.

scott-huberty, 2023-10-25:
> Yes but let's be honest, we're constantly breaking user code and the short deprecation cycles we currently have are more an alibi to soothe ourselves (the devs) than really helping users actively using our software.

Just wanted to chime in that as a _user_ I've experienced this. For example in [plot_topomap](https://mne.tools/stable/generated/mne.viz.plot_topomap.html) the `show_names` parameter was deprecated in `1.2` and switched to `names` in `1.3`, IIRC the deprecation cycle was like 2 months so I think I just missed it entirely and only noticed when my code for a project broke after updating. It was a super easy fix (for me) but I remember having to check the doc across versions to figure out what happened.

But maybe that's a digression from this ticket. Anyways I think that as a _user_, some mechanism in `sys_info` that communicates that my version is outdated would probably come in handy.

cbrnr, 2023-10-25:
I like the idea to include a message in the output of `mne.sys_info()`, no warnings or depreciations necessary.

hoechenberger, 2023-10-25:
I also like the idea of adding that information to sys_info(). But simply as an info and not immediately as a nudge to upgrade 😃

larsoner, 2023-10-26:
PR opened in #12146, feedback welcome

closed 2023-10-30 by drammock

cbrnr, 2023-11-02:
> [...] one of my pet peeves with R is the ~7 lines of output it spits on your console every time you fire it up

@drammock there are two ways you can suppress this message:

1. Start R with the `-q` flag.
2. Unfortunately, RStudio ignores all startup flags. Therefore, the first option doesn't work, but you can work around this problem by creating `~/.Rprofile` with the following content:

   ```r
   system2("clear")
   cat("\014")
   ```