mne-tools/mne-python #12134: BrainVision IO -- Comment section is partially hardcoded and may lead to problems

sappelhoff, 2023-10-25:
For example here we assume that if a "Channels" section is present in the "Comment" section, it always has a certain number of "columns":

https://github.com/mne-tools/mne-python/blob/4518a1a918051542ab7360ff63c529121a499417/mne/io/brainvision/brainvision.py#L674C24-L674C30

But this is not necessarily the case, see:

https://mne.discourse.group/t/indexerror-in-brainvision-py/7187

I think we should amend our code to do a quick parsing of the number of columns in the "Channels" section if it is present.

cbrnr, 2023-10-25:
What does the standard say?

sappelhoff, 2023-10-26:
The standard says:

[image]

source: https://www.brainproducts.com/support-resources/brainvision-core-data-format-1-0/ (and the linked  PDF there)

cbrnr, 2023-10-26:
Ah OK then 😄!

closed 2026-11-10 by larsoner