mne-tools/mne-python #12137: Should PSD plots in the report contain bad channels by default?

hoechenberger, 2023-10-25:
Since we switched from plot_psd() to compute_psd().plot(), PSD plots in the Report contain bad channels. Is this a default we want to keep? I was using MNE-BIDS-Pipeline and triple-checked my data until I realized what's going on. (I thought something is wrong with my BIDS conversion) 
So even if we keep this behavior for Report, we should reconsider it for MNE-BIDS-Pipeline (and transfer this issue if needed)

hoechenberger, 2023-10-25:
One argument for excluding bads is that the PSDs are added for Raw and Epochs instances, where the data traces are currently not corresponding to the PSD since the former do jot contain bads

drammock, 2023-10-25:
> Since we switched from plot_psd() to compute_psd().plot(), PSD plots in the Report contain bad channels

actually the change was here: https://github.com/mne-tools/mne-python/pull/11680 and the motivation was a dev-meeting discussion where I think @agramfort made the point that PSDs are often used for data quality checks and thus showing bads is a useful default to have.

> Is this a default we want to keep?

I think so. And after #12006 it should now be possible to specify `exclude="bads"` either when creating the spectrum or when plotting.

hoechenberger, 2023-10-26:
Yes, I'm using the PSD plots myself for data quality checks and I can see that for interactive use, showing the bads by default makes sense.

But for the reports, the data has usually already been inspected or processed by the time it gets added to the report. Now the issue is that the bad channels are not visually distinct from the non-bads, and since the PSD plot in the report is not interactive, there is no way to figure out (post-hoc, when looking at the report) whether a channel with an odd PSD has actually been marked as bad or not.

This is especially problematic when using MNE-BIDS-Pipeline, where one cannot be sure which channels actually entered analysis…

drammock, 2023-10-26:
> * in Report, don't show bads in PSD by default for `add_raw()` and `add_epochs()`

This is the approach I would take. Sorry, I think I misunderstood your earlier post as suggesting to change the method default, not just the default within report contexts

larsoner, 2023-10-26:
Changing the `report.add_raw/epochs(..., psd=True)` behavior sounds reasonable to me. But I think really some variant of this would be the best solution:

> plot PSD traces of bad channels in a different color so they're easy to spot

Given that we use `spatial_colors` (and that this is useful) I'm not totally sure what would work. One option would be to use middle gray since it would stand out as the only `saturation=0` color. (And in cases where spatial colors are not used then we can use red as usual, which I think is what `evoked.plot(spatial_colors=False)` does for example.) Maybe we should look to see how this looks with `evoked.plot()` butterfly for example where `spatial_colors=True`. I don't actually know what we do for a bad color there (?) but gray might be easy enough to spot as bad in the middle of the channel rainbow.

hoechenberger, 2023-10-26:
@larsoner
If finding a suitable color turns out to be a problem, we could consider selecting a different line style (dashed?) maybe?

larsoner, 2023-10-26:
I thought about this but it can be really hard to follow dotted or dashed lines when the lines are squiggly, and evoked/psd lines can be quite squiggly

hoechenberger, 2023-10-26:
In that case – maybe a very red red? :) I don't think we use that as a spatial color, do we?

larsoner, 2023-10-26:
> In that case – maybe a very red red? :)

We use a whole bunch of colors, including red. :( I think it would be hard to find a high-saturation color that stands out, hence the idea to use a color with saturation zero.

I quickly modified `mne/viz/evoked.py` to try middle gray. Looks like we might already be setting the line style in `evoked.plot` to be an acceptable dashing, too (I didn't set it), but I think the middle gray also helps distinguish it further:

<details>
<summary>Patch</summary>

```
diff --git a/mne/viz/evoked.py b/mne/viz/evoked.py
index 6abcbcc0d1..1c6712a6be 100644
--- a/mne/viz/evoked.py
+++ b/mne/viz/evoked.py
@@ -679,15 +679,17 @@ def _plot_lines(
                     _handle_spatial_colors(
                         colors, info, idx, this_type, psd, ax, sphere
                     )
+                    bad_color = (0.5, 0.5, 0.5)
                 else:
                     if isinstance(_spat_col, (tuple, str)):
                         col = [_spat_col]
                     else:
                         col = ["k"]
+                    bad_color = "r"
                     colors = col * len(idx)
-                    for i in bad_ch_idx:
-                        if i in idx:
-                            colors[idx.index(i)] = "r"
+                for i in bad_ch_idx:
+                    if i in idx:
+                        colors[idx.index(i)] = bad_color
 
                 if zorder == "std":
                     # find the channels with the least activity

```

</details>

```
import mne
evoked = mne.read_evokeds(mne.datasets.sample.data_path() / "MEG" / "sample" / "sample_audvis-ave.fif", baseline=(None, 0))[0]
evoked.pick("eeg", exclude=())
evoked.plot(spatial_colors=True, exclude=())
```
[image]

Zoomed:

[image]

And here is spatial_colors=False:

[image]

larsoner, 2023-10-26:
Sorry I mean "stands out" as in "is easy to tell that it's different/separate from the good, spatial-color-mapped channels", not "pops out as prominent". Just that we need them to be differentiable in some way

hoechenberger, 2023-10-26:
> Sorry I mean "stands out" as in "is easy to tell that it's different/separate from the good, spatial-color-mapped channels", not "pops out as prominent". Just that we need them to be differentiable in some way

Yes, that's how I understood it :) I meant to say, maybe gray is a good choice after all, even it it's not easily discoverable if the traces for all channels are close together. Because then, the bad channel won't raise any eyebrows. Currently, it **does** raise eyebrows in my analysis because the trace is spatially displaced from the other traces, **and** it has the regular spatial color, so it's not immediately obvious that it's bad. If it were gray, it would already help

PR #12142 (2023-10-27): ENH: set color for bad channel with spatial_colors=True
Closes #12137

@hoechenberger can you try it with your data to see if it helps?

This will affect at least `evoked.plot` and `spectrum.plot` as `spatial_colors=True` by default.

- [x] See if it actually helps with reports
- [x] Add test
- [x] Add changelog entry
closed 2023-10-27 by larsoner