mne-tools/mne-python #12144: We may soon be able to drop all mamba-related info from our install docs

hoechenberger, 2023-10-26:
### Proposed documentation enhancement

The mamba solver will become the default solver for conda soon, hence no need for ordinary users to explicitly touch mamba anymore:

> The upcoming, special 23.10.0 release will be dedicated to the switch of the default solver to libmamba.

https://github.com/conda/conda/releases/tag/23.9.0

drammock, 2023-10-26:
It will be nice to purge that language!  I think we should still say something like "if you have problems with the "solving environment" step failing or being slow, make sure you're on conda version 23.10.0 or newer".

hoechenberger, 2023-12-11:
Mamba is / will be deprecated as the solver is now fully integrated into `conda`. See this PR:
https://github.com/mamba-org/mamba/pull/3054

This means that we should recommend users install `"conda >=23.10"` and remove all references to `mamba` from the docs.

For CI, we should switch from Mambaforge to Miniforge.

cc @larsoner @cbrnr

AntoinePrv, 2023-12-18:
> The mamba CLI is / will be deprecated as the solver is now fully integrated into conda. See this PR:

That's not quite what is happening. Mamba is still a first class citizen. What is happening is `mamba.py` which was glue between conda and libmamba has been removed in favor of a full C++ CLI client.

jonashaag, 2023-12-18:
@AntoinePrv since the Micromamba CLI is not compatible with Mamba, we _are_ in fact deprecating Mamba.

I think what you call "mamba.py" in your comment is typically called "Mamba", and what you call "Mamba" is typically called "libmamba". IMO we should stick to the following terminology:

- Mamba: The Conda-compatible Python CLI that will be deprecated.
- Micromamba: The C++ binary based on libmamba.
- libmamba: The C++ library.
- libmambapy: The Python binding to the C++ library.

With this terminology, "Mamba" will be deprecated.

hoechenberger, 2023-12-18:
Yes this is what I meant! The mamba binary is deprecated. The mamba solver library and micromamba binary are not. Thanks for the clarification, @jonashaag

AntoinePrv, 2023-12-18:
Yes, except that we will ship a `mamba` executable that uses the same code as `micromama`. The name will remain.

hoechenberger, 2023-12-18:
Ah okay, this is good to know!

AntoinePrv, 2023-12-18:
@0xbe7a as in `ln -s micromamba mamba` (and micromamba is a conda-compatible frontend, we are interested in pairing any reported missing feature)

larsoner, 2024-07-15:
Based on `git grep` I think we're good here already!

PR #12121 (2023-11-09): Fix copy-view issue in epochs
Closes #12144
Closes #12114

- [x] Add deprecation
- [x] Get pytest runs green
- [x] Get `[circle full]` green
closed 2024-07-15 by larsoner