mne-tools/mne-python #12147: JS inspector error

larsoner, 2023-10-26:
JS inspector in Safari on macOS says:

```
[Error] ReferenceError: Can't find variable: $
	Global Code (30_info.html:1323)
```

The respective line is:
```html
<script type="text/javascript">$(function () { $("time.localized").each(function () { var el = $(this); el.text(new Date(el.attr("datetime")).toLocaleString([], {dateStyle: "medium", timeStyle: "long"})); }); } )</script></p></div>
```

not sure if related, though!

but it seems as if jQuery is expected but not embedded

_Originally posted by @hoechenberger in https://github.com/mne-tools/mne-python/issues/12130#issuecomment-1781572448_

larsoner, 2023-10-26:
At some point we added some jQuery magic to convert something to a user's local timezone. I think it was the "last updated" stuff. Now that pydata-sphinx-theme no longer uses jQuery (I think?) we should scrap it or fix it. In the meantime the result for end-users is the same as if we just scrapped it I think.

drammock, 2023-10-26:
> Now that pydata-sphinx-theme no longer uses jQuery (I think?)

yes indeed PST purged JQuery a while back. Let's translate to vanilla JS.