mne-tools/mne-python #12152: Report.save() ignores `verbose` parameter value

hoechenberger, 2023-10-30:
### Description of the problem

Saving a report and setting `verbose=False`, `verbose="warning"`, or `verbose="error"` still prints out log message of level `info`.

### Steps to reproduce

```Python
import mne

r = mne.Report()
r.save("/tmp/foo.html", overwrite=True, open_browser=False, verbose="error")
```

### Expected results

No output

### Actual results

```
Embedding : jquery-3.6.0.min.js
Embedding : bootstrap.bundle.min.js
Embedding : bootstrap.min.css
Embedding : bootstrap-table/bootstrap-table.min.js
Embedding : bootstrap-table/bootstrap-table.min.css
Embedding : bootstrap-table/bootstrap-table-copy-rows.min.js
Embedding : bootstrap-table/bootstrap-table-export.min.js
Embedding : bootstrap-table/tableExport.min.js
Embedding : bootstrap-icons/bootstrap-icons.mne.min.css
Embedding : highlightjs/highlight.min.js
Embedding : highlightjs/atom-one-dark-reasonable.min.css
```

### Additional information

–

larsoner, 2023-10-30:
I think this is a usage error, you need `verbose=False` in the constructor as that's when the embedding happens:
```
>>> r = mne.Report()
Embedding : jquery-3.6.0.min.js
Embedding : bootstrap.bundle.min.js
Embedding : bootstrap.min.css
Embedding : bootstrap-table/bootstrap-table.min.js
Embedding : bootstrap-table/bootstrap-table.min.css
Embedding : bootstrap-table/bootstrap-table-copy-rows.min.js
Embedding : bootstrap-table/bootstrap-table-export.min.js
Embedding : bootstrap-table/tableExport.min.js
Embedding : bootstrap-icons/bootstrap-icons.mne.min.css
Embedding : highlightjs/highlight.min.js
Embedding : highlightjs/atom-one-dark-reasonable.min.css
>>> r = mne.Report(verbose=False)
>>>
```

closed 2023-10-30 by larsoner

hoechenberger, 2023-10-30:
But `Report.save()` does have a `verbose` parameter, why can I not control things via that parameter directly?

hoechenberger, 2023-10-30:
Oh. Now I see what you mean! The output is not generated during saving, but when the Report is being instantiated! Got it!!