mne-tools/mne-python #12153: Cannot concat epochs if baseline period was cropped off

hoechenberger, 2023-10-30:
### Description of the problem

`concatenate_epochs()` refuses to operate on Epochs if their baseline period has been cropped away.

### Steps to reproduce

```Python
# %%
import mne

sample_dir = mne.datasets.sample.data_path()
sample_fname = sample_dir / "MEG" / "sample" / "sample_audvis_raw.fif"

raw = mne.io.read_raw_fif(sample_fname)
raw.crop(tmax=60)

events = mne.find_events(raw, stim_channel="STI 014")
event_id = {
    "auditory/left": 1,
    "auditory/right": 2,
    "visual/left": 3,
    "visual/right": 4,
    "face": 5,
    "buttonpress": 32,
}

epochs = mne.Epochs(
    raw,
    events=events,
    event_id=event_id,
    tmin=-0.2,
    tmax=0.5,
    baseline=(None, 0),
    preload=True,
)

epochs.crop(0, None)  # crop away baseline period
mne.concatenate_epochs([epochs, epochs])
```

### Expected results

Concatenation should work

### Actual results

```python
ValueError: Baseline interval [-0.19979521315838786, 0.0] s is outside of epochs data [0.0, 0.49948803289596966] s. Epochs were probably cropped.
```

### Additional information

–

agramfort, 2023-10-30:
why cropping the baseline in the first place?

> Message ID: ***@***.***>
>

hoechenberger, 2023-10-30:
> why cropping the baseline in the first place?

It's something we do in MNE-BIDS-Pipeline during decoding, which we usually start at time point zero (as to omit the baseline). We crop the epochs and then concatenate epochs of different conditions before feeding them to the classifier.

I have now simply changed the order of doing things: first concat, then crop.

It's a viable workaround, but I believe not being able to contact if the baseline has been cropped is unintended behavior, hence a bug

Nothing urgent, but I think we should fix it if / when we have time

The problem started showing up while working on https://github.com/mne-tools/mne-bids-pipeline/pull/796

agramfort, 2023-11-01:
I would argue that we should decode also during baseline to find if there
are issues in the baseline. During
baseline the decoder should be at chance level

> Message ID: ***@***.***>
>

hoechenberger, 2023-11-01:
@agramfort I see we offer a config option for that:
https://mne.tools/mne-bids-pipeline/1.4/settings/sensor/mvpa.html#mne_bids_pipeline._config.decoding_epochs_tmin

We default to starting decoding at t=0, but we do support t=None for the beginning of the epochs. Do you think we should change the default?

hoechenberger, 2023-11-01:
@agramfort I misunderstood that setting. It only applies to **full epochs decoding**, where I believe our default (starting with t=0) makes sense

For time-by-time decoding, we already decode across the entire epochs range, including the baseline, and here's no config setting to change this.

So I think we should be good as far as implementation in MNE-BIDS-Pipeline goes

agramfort, 2023-11-01:
Yes then we’re good 👍