mne-tools/mne-python #12158: Small error in interpolation tutorial

hoechenberger, 2023-11-01:
### Proposed documentation enhancement

https://mne.tools/dev/auto_tutorials/preprocessing/15_handling_bad_channels.html#interpolation-in-mne-python

Says:

> Note that we used the exclude=[] trick in the call to [pick_types()](https://mne.tools/dev/generated/mne.io.Raw.html#mne.io.Raw.pick_types) to make sure the bad channels were not automatically dropped from the selection.

But `pick_types()` isn't used anymore (we switched to `inst.pick()`)

btkcodedev, 2023-12-13:
@hoechenberger  i am a newbie for this repo, could i work on this issue :)

hoechenberger, 2023-12-13:
Absolutely! Feel free to ask if you have any questions 👌

btkcodedev, 2023-12-26:
Hi, @hoechenberger, I've raised MR https://github.com/mne-tools/mne-python/pull/12326

PR #12326 (2024-01-25): chore(docs): update preprocessing tutorial for using inst.pick() instead of pick_types()
#### Reference issue
Closes #12158

Updated the tutorial by removing legacy method call of mne.io.Raw.pick_types() to instance.pick()
closed 2024-01-25 by mscheltienne