mne-tools/mne-python #12159: Once more: Traces of bad channels in PSD figures

hoechenberger, 2023-11-01:
#12142 improved the discoverability of bad channels in PSD plots. However, I have to say I'm sometimes still struggling to really tell these traces apart from the non-bass; maybe it's the slightly blurred figures we add to Reports; maybe it's my display; maybe it's my eyes. Here's an example:

[image]

I cannot really see that the line is dashed / pointed, unless I look super super closely. In fact, at first I panicked and re-analyzed that dataset over and over again, trying to find that one rogue channel, until I discovered it had already been marked as bad. I just didn't really see it.

Any ideas on how things could be further improved would be greatly appreciated!

drammock, 2023-11-01:
a couple ideas:

1. add the names of bad channels to the title, subtitle, or (specifically in reports) to the text below the plot (that currently says "PSD calculated from..."
2. add a hover effect to the traces and make it so that on hover, bad channel labels say e.g., "MEG 1332 (BAD)"
3. in reports, don't plot with `spatial_colors=True` (so the good traces are black and the bad ones are red)
4. make bad line thicker. In my experience dash/dot patterns are easier to see on thicker lines

larsoner, 2023-11-01:
Another option along the lines of (4) is to adjust the dashing on the bad line to be more spaced out. I think a big part of the problem is that it's very difficult to see the dot/dashed-ness of the line when the linewidth is 0.5 even at full res. Once you scale it down it pretty much goes away. I think I prefer this and/or increasing the line thickness (4) over the other options