mne-tools/mne-python #12169: Editable installs with pip and VS Code

hoechenberger, 2023-11-03:
Since we adopted PEP 660 by moving to `pyproject.toml`, I'm having trouble with Pylance (VS Code) when working with editable installs.

Usually I do:

```shell
pip install -e .
```
from the MNE-Python directory. I can then import`mne`, and Pylance will recognize the package and provide tab completion etc.

Since we adopted `pyproject.toml`, I had to install in non-editable mode to have Pylance recognize the package.

I started digging a little and found out it's a [pain point](https://github.com/microsoft/pylance-release/issues/3473#issuecomment-1619179711) for static type checkers (like Pylance) introduced through PEP 660.

The [proposed solution](https://github.com/microsoft/pylance-release/blob/main/TROUBLESHOOTING.md#editable-install-modules-not-found) is to run `setuptools` in "legacy"  or "compat" mode. However, [recent releases of `pip` don't support this anymore](https://setuptools.pypa.io/en/latest/userguide/development_mode.html#legacy-behavior).

My current solution is using the new ["strict" mode](https://setuptools.pypa.io/en/latest/userguide/development_mode.html#strict-editable-installs):
```shell
pip install --config-settings editable-mode=strict -e .
```
With that, everything works as expected, with the only limitation that newly created files don't show up in the editable install – one has to "re-install" in that case. But I can live with that.

We should probably document this behavior.

Another alternative is switching our build backend to [Hatch](https://hatch.pypa.io/latest/), like Napari did [recently](https://github.com/napari/npe2/issues/280). But this step seems to be a bit much.

cbrnr, 2023-11-03:
Wait, I've been using editable installs and Visual Studio Code forever, and I haven't noticed any regressions recently. Can you give an example what's not working anymore?

hoechenberger, 2023-11-03:
I've had this issue for months, it only occurs with new versions of setuptools and pip

```shell
❯ mamba create -c conda-forge -n mne-base-test mne-base
❯ mamba activate mne-base-test
❯ mamba uninstall --force mne-base ipykernel
❯ pip --version
pip 23.3.1 from /Users/richardhochenberger/mambaforge/envs/mne-base-test/lib/python3.12/site-packages/pip (python 3.12)
❯ pip install --quiet -e ~/Development/mne-python
❯ code foo.py
```
Ensure it's using the correct environment

Add the following content to the file:
```python
# %%
import mne
```
Save the file

You will get red squiggly lines under `mne`:

> Import "mne" could not be resolvedPylance[reportMissingImports](https://github.com/microsoft/pyright/blob/main/docs/configuration.md#reportMissingImports)

Close VS Code

Run:

```shell
❯ pip install --quiet --config-settings editable-mode=strict -e ~/Development/mne-python
```

Start VS Code:
```shell
❯ code foo.py
```

Now Pylance doesn't complain about the missing import anymore.

- Can only be replicated if using new (the latest?) versions of setuptools and pip
- When trying to replicate, ensure that you're **not** inside the `mne-python` directory, as Pylance may pick up the nested `mne` directory there (I'm not 100% sure about this but it's better to exclude this possibility)

cbrnr, 2023-11-03:
I can confirm that this is indeed not working. However, as you suspected, it *does* work if you your working directory is actually `mne-python`. So for developing MNE-Python, the editable install works as expected.

The solutions/workarounds you pointed out don't really sound convincing to me. Why do you need an editable install? If you have to re-install even in editable strict mode when adding new files, this seems a bit pointless (meaning you might as well install in non-editable mode and re-install every time you make a change). What exactly is your use case?

I'm fine with documenting this special case, but as we've mentioned previously, this does not affect when you're developing MNE-Python, so I'm not even sure where we should document it.

hoechenberger, 2023-11-03:
> If you have to re-install even in editable strict mode when adding new files, this seems a bit pointless (meaning you might as well install in non-editable mode and re-install every time you make a change).

Normally I edit existing files only; I don't add new ones. Changes in files that existed by the time the "strict" editable install was performed will be reflected when importing the package. Only newly-created files don't show up.

> What exactly is your use case?

For example, when I'm developing MNE-BIDS-Pipeline and want to improve `mne.Report`, I have my workspace / environment MNE-BIDS-Pipeline and and editable install of MNE-Python, such that every time I change something on my MNE-Python development branch, the Pipeline will pick it up upon a re-run.

I had a similar issue recently when developing MNE-BIDS.

hoechenberger, 2023-11-03:
Excellent resource, @cbrnr!

@drammock I can look into this tomorrow

hoechenberger, 2023-11-03:
Just a note to self, I just ran into the same issue when trying to use an editable install of MNE-BIDS.

With `autoreject`, I receive:

```shell
❯ pip install --quiet --config-settings editable-mode=strict -e ../autoreject
  DEPRECATION: Config settings are ignored for project file:///Users/richardhochenberger/Library/CloudStorage/OneDrive-FirmenichSA/Documents/Development/autoreject. pip 24.0 will enforce this behaviour change. A possible replacement is to use --use-pep517 or add a pyproject.toml file to the project.
```

cc @sappelhoff

cbrnr, 2023-11-03:
I'd be interested in what adopting `hatchling` as a build backend means for users. It will probably not affect normal `pip install mne`, right? But I'm assuming `pip install -e /some/path/to/mne-python` will require the `hatch` package. Is this correct?

hoechenberger, 2023-11-03:
@cbrnr The nice thing about `pyproject.toml` and PEP 660 (?) is that we can specify a build system:

https://github.com/mne-tools/mne-python/blob/fcb59266b7616dd99908c6dac0e2295722b0c567/pyproject.toml#L45-L47

This will be installed before the build process starts.

cbrnr, 2023-11-03:
But what is "the build process"?

hoechenberger, 2023-11-03:
> But what is "the build process"?

The package is first built, then installed, when you run `pip install ./my-package`

cbrnr, 2023-11-03:
So it's the process of building a wheel, more or less, right? Which means that this change does not affect `pip install mne` at all, because the wheel has already been built.

drammock, 2023-11-03:
> One thing we'd need to watch out for is whether we can still get `setuptools_scm`-like behavior for our version string.

Looks like that won't be a problem... from the scientific-python page that @cbrnr linked to:

> You can tell hatchling to get the version from VCS. Add `hatch-vcs` to your `build-backend.requires`, then add the following configuration:
>
> ```
> [tool.hatch]
> version.source = "vcs"
> build.hooks.vcs.version-file = "src/<package>/version.py"
> ```

hoechenberger, 2023-11-03:
> So it's the process of building a wheel, more or less, right?

Yes

ofek, 2023-12-03:
Maintainer of Hatch/Hatchling here! Let me know if I can be of assistance 🙂

PR #12269 (2023-12-08): Use hatchling as build backend
Supersedes #12173

Closes #12173 
Closes #12169

cc @ofek Let's see how this goes

Editable installs now work as expected.

------

I built the packages via
```shell
python -m build .
```
both on this branch and on `main`.

I diffed the contents of the archives.

### Wheel diff
```shell
❯ diff setuptools/site-packages hatchling/site-packages
Common subdirectories: setuptools/site-packages/mne and hatchling/site-packages/mne
Only in setuptools/site-packages: mne-1.7.0.dev26+g7bf1b4ab7.dist-info
Only in hatchling/site-packages: mne-1.7.0.dev27+ge9f4a7be3.dist-info
 [... cut]
closed 2023-12-08 by hoechenberger