mne-tools/mne-python #12180: Add Star History to readme?

hoechenberger, 2023-11-07:
### Proposed documentation enhancement

I just came across this:
https://star-history.com/#mne-tools/mne-python&Date

I think it could be fun to showcase in our readme. WDYT?

[image]

cbrnr, 2023-11-07:
Looks fun, but without comparing to another (similar) project, it isn't really that informative. Or why would anyone (except us devs) be interested in the time course of our stars?

hoechenberger, 2023-11-07:
I thought it shows continuing popularity :)

drammock, 2023-11-07:
to me this conveys information more similar to what is on the "insights" tab of the repo. I don't think it's appropriate for the README

closed 2023-11-07 by hoechenberger