mne-tools/mne-python #12191: plot_compare_evokeds doesn't display title when using axes='topo'

withmywoessner, 2023-11-09:
### Description of the problem

plot_compare_evokeds doesn't display title when using axes='topo'
Example: mne.viz.plot_compare_evokeds(evk, picks="eeg", axes="topo", title='TEST ERP')
Maybe this is not a bug and this was never intended. If so, maybe documentation should be updated?

### Steps to reproduce

```Python
sample_data_folder = mne.datasets.sample.data_path()
sample_data_raw_file = (
    sample_data_folder / "MEG" / "sample" / "sample_audvis_filt-0-40_raw.fif"
)

raw = mne.io.read_raw_fif(sample_data_raw_file)
events = mne.find_events(raw, stim_channel="STI 014")
event_dict = {
    "auditory/left": 1,
    "auditory/right": 2,
    "visual/left": 3,
    "visual/right": 4,
    "smiley": 5,
    "buttonpress": 32,
}
reject_criteria = dict(
    mag=4000e-15,  # 4000 fT
    grad=4000e-13,  # 4000 fT/cm
    eeg=150e-6,  # 150 µV
    eog=250e-6,
)  # 250 µV
epochs = mne.Epochs(
    raw,
    events,
    event_id=event_dict,
    tmin=-0.2,
    tmax=0.5,
    reject=reject_criteria,
    preload=True,
)
evk= epochs.average()
mne.viz.plot_compare_evokeds(evk, picks="eeg", axes="topo", title='TEST ERP')
```

### Expected results

Plot should have 'TEST ERP' at the top of the figure. 

### Actual results

[image]

### Additional information

Platform             Linux-5.15.133.1-microsoft-standard-WSL2-x86_64-with-glibc2.31
Python               3.11.6 | packaged by conda-forge | (main, Oct  3 2023, 10:40:35) [GCC 12.3.0]
Executable           /home/woess/mambaforge/envs/mne/bin/python
CPU                  x86_64 (12 cores)
Memory               11.3 GB

Core
├☑ mne               1.5.1
├☑ numpy             1.24.4 (OpenBLAS 0.3.24 with 12 threads)
├☑ scipy             1.11.3
├☑ matplotlib        3.8.0 (backend=module://matplotlib_inline.backend_inline)
├☑ pooch             1.7.0
└☑ jinja2            3.1.2

Numerical (optional)
├☑ sklearn           1.3.1
├☑ numba             0.57.1
├☑ nibabel           5.1.0
├☑ nilearn           0.10.2
├☑ dipy              1.7.0
├☑ openmeeg          2.5.6
├☑ pandas            2.1.1
└☐ unavailable       cupy

Visualization (optional)
├☑ pyvista           0.42.3 (OpenGL 3.3 (Core Profile) Mesa 21.2.6 via D3D12 (AMD Radeon(TM) Graphics))
├☑ pyvistaqt         0.0.0
├☑ vtk               9.2.6
├☑ qtpy              2.4.0 (PyQt5=5.15.8)
├☑ pyqtgraph         0.13.3
├☑ mne-qt-browser    0.5.2
├☑ ipywidgets        8.1.1
├☑ trame_client      2.12.5
├☑ trame_server      2.12.0
├☑ trame_vtk         2.5.9
├☑ trame_vuetify     2.3.1
└☐ unavailable       ipympl

Ecosystem (optional)
└☐ unavailable       mne-bids, mne-nirs, mne-features, mne-connectivity, mne-icalabel, mne-bids-pipeline

larsoner, 2023-11-10:
I agree it's arguably a bug. Can you see if `fig.suptitle` is good enough here? If so then we can fix the bug. If it looks bad, then maybe we just update the docs to say title is ignored in topo mode.

withmywoessner, 2023-11-10:
It worked @larsoner. I just made some random temporary variables because the values of title and axes change at some point. Making a PR to discuss formatting.

PR #12192 (2023-11-14): [WIP] [BUG] Display title in plot_compare_evokeds when axes='topo'

#### Reference issue
Example: Fixes #12191.

#### What does this implement/fix?
Add title to topoplot in `plot_compare_evokeds`

#### Additional information
Any additional information you think is important.
closed 2023-11-14 by larsoner