mne-tools/mne-python #12193: Read raw data from the Curry company

PhD-GOAT, 2023-11-11:
### Description of the problem

It is impossible to read the cef file, which is Curry's raw data.

### Steps to reproduce

```Python
import mne

raw = mne.io.read_raw_curry('test.cef')
```

### Link to data

[image]

### Expected results

Expected to return an instance of Raw object.

### Actual results

FileNotFoundError: The following required files cannot be found: ['data' 'info' 'labels'].
Please make sure all required files are located in the same directory as test.hpi.

### Additional information

Platform             Windows-10-10.0.19045-SP0
Python               3.10.9 (tags/v3.10.9:1dd9be6, Dec  6 2022, 20:01:21) [MSC v.1934 64 bit (AMD64)]
Executable           C:\AppData\Local\Programs\Python\Python310\python.exe
CPU                  Intel64 Family 6 Model 151 Stepping 5, GenuineIntel (12 cores)
Memory               31.8 GB

Core
├☑ mne               1.4.2
├☑ numpy             1.25.2 (OpenBLAS 0.3.23.dev with 12 threads)
├☑ scipy             1.11.2
├☑ matplotlib        3.7.2 (backend=module://matplotlib_inline.backend_inline)
├☑ pooch             1.6.0
└☑ jinja2            3.1.2

Numerical (optional)
├☑ sklearn           1.2.2
├☑ pandas            2.1.0
└☐ unavailable       numba, nibabel, nilearn, dipy, openmeeg, cupy

Visualization (optional)
└☐ unavailable       pyvista, pyvistaqt, ipyvtklink, vtk, qtpy, ipympl, pyqtgraph, mne-qt-browser

Ecosystem (optional)
└☐ unavailable       mne-bids, mne-nirs, mne-features, mne-connectivity, mne-icalabel, mne-bids-pipeline

agramfort, 2023-11-11:
sorry but we cannot help without more details and access to the data to
replicate the issue.

> Message ID: ***@***.***>
>

wmvanvliet, 2023-11-13:
going off the error message, are the data, info and label files there alongside the `.cef` file?

PhD-GOAT, 2023-11-13:
> going off the error message, are the data, info and label files there alongside the `.cef` file?

I am trying with only one cef file and it is requesting hpi file. In the first place, aren't labels needed unless process them in an annotation function? The cef file contains only number list values.

wmvanvliet, 2023-11-13:
yes, this is what the error message is trying to tell you. I think the "read_raw_curry" function tries to find the "sidecar" files that contain the labels and such automatically, cannot find them, and hence gives this error. These files probably need to be in the same folder as the `.cef` file