mne-tools/mne-python #12196: Add functionality of Epochs class to make_fixed_length_epochs

withmywoessner, 2023-11-12:
### Describe the new feature or enhancement

Hello! I was looking over the `make_fixed_length_epochs` function, and I realized that it is missing a lot of the functionality of the `Epochs `class such as `picks`, `reject `and `flat`. I understand that fixed length epochs can easily be created using `mne.make_fixed_length_events` and then passing the events to the `Epochs` constructor, but I was wondering if it would be useful to add some of that functionality to the `make_fixed_length_epochs` function. Maybe its not that useful and is fine as is. 

### Describe your proposed implementation

Adding new parameters and functions to match functionality of `Epochs` class.

### Describe possible alternatives

Maybe add a note in the documentation that the preferred method of creating fixed length epochs is `mne.make_fixed_length_events`  then calling the `Epochs` constructor so that if more features are added to `Epochs`, users won't be left behind.

agramfort, 2023-11-12:
you can see do it after with Epochs.drop_bad(reject=...).pick(...) no?

> Message ID: ***@***.***>
>

withmywoessner, 2023-11-13:
Yes, you can @agramfort. I just thought it would be an easy change to make the parameters more consistent with epochs as the function just calls the Epoch constructor:
```
events = make_fixed_length_events(raw, id=id, duration=duration, overlap=overlap)
    delta = 1.0 / raw.info["sfreq"]
    return Epochs(
        raw,
        events,
        event_id=[id],
        tmin=0,
        tmax=duration - delta,
        baseline=None,
        preload=preload,
        reject_by_annotation=reject_by_annotation,
        proj=proj,
        verbose=verbose,
    )

```
If its unnecessary that's fine too!

larsoner, 2023-11-13:
I'm inclined not to add more stuff to `make_fixed_length_epochs`. If you need full flexibility there is always `Epochs(..., make_fixed_length_events(...), ...)`. Otherwise anything we change in `Epochs` we need to add to `make_fixed_length_epochs` which would be nice to avoid...

withmywoessner, 2023-11-13:
Okay, sounds good. I will close this then.

closed 2023-11-13 by withmywoessner