mne-tools/mne-python #12199: EEG gradient artefact correction pre-processing function

sfc-neuro, 2023-11-13:
### Describe the new feature or enhancement

A preprocessing function able to correct the gradient artefact on EEG data recorded simultaneously as fMRI. 

### Describe your proposed implementation

Code implementation of the moving average filter described in Ferreira, José L et al. (2016).

Ref:
Ferreira, José L et al. “Gradient Artefact Correction and Evaluation of the EEG Recorded Simultaneously with fMRI Data Using Optimised Moving-Average.” Journal of medical engineering vol. 2016 (2016): 9614323. doi:10.1155/2016/9614323

### Describe possible alternatives

Possibly translating the FASTR function from the fMRIB matlab package that already performs this preprocessing would be a viable option.

Ref:
 Improved FMRI Artifact Reduction from Simultaneously Acquired EEG Data using Slice Dependant Template Matching, Rami K. Niazy, Giandomenico Iannetti, Christian Beckmann, Michael Brady, Stephen Smith. Proc. 12th ISMRM (2004), Kyoto, Japan

larsoner, 2023-11-13:
If FASTR is better then it would be nice to have that. Is the code under a BSD compatible license?

wmvanvliet, 2023-11-13:
I think in theory, the OMA filter is better as it has less finicky parameters to set. However, we're currently trying to deal with filter ringing artifacts at the boundaries of the data.

FASTR is good because there's a reference implementation for it. The bad thing about it is that it has a lot of parameters that the user must get exactly right.

larsoner, 2023-11-13:
Okay I am naive on this front, happy to let you decide @wmvanvliet !

pmolfese, 2023-12-27:
We wrote a function for doing essentially this, although my memory fails me if it was FASTR or not. 

https://github.com/nimh-sfim/gradient_remover

The main hiccup for us was that some of the import functions (looking at EGI in particular) don't give exact timing and so building the template for removal of these artifacts was difficult. Our function takes the exported EGI event timing to pull this off, but the ideal would be to make sure that every importer either 1) correctly imports timing exactly or 2) keeps a copy of the accessory times to try and sort this out.

vferat, 2024-01-11:
There is also an online implementation on neuxus: [here](https://github.com/LaSEEB/NeuXus/blob/ed5d8d5a588c6346353b92e4e9fb276ba648e04a/neuxus/nodes/correct.py#L15) (seems a bit less advanced than @pmolfese code)

I could be able to provide EO/EC recordings with scan on/off if needed.