mne-tools/mne-python #12217: Web-based EEG/MEG browser/viewer

cbrnr, 2023-11-16:
I've recently been pointed to #12030, where a web-based viewer for 3D brain images is suggested (great idea BTW). In related news, Anaconda is currently working on implementing a time series browser/viewer specifically for EEG/MEG signals. I've been in contact with @droumis from the beginning of the project to make sure that it could be a viable alternative backend for our raw browser.

If you are interested, this is the main project website: https://github.com/holoviz-topics/neuro

The relevant workflow for us is Stacked Timeseries: https://github.com/holoviz-topics/neuro/blob/main/workflows/stacked-timeseries/0-StackedTimeseries.ipynb

The project will be running for 12 more months, so if you have specific feedback and/or feature requests, please feel free to drop them here. Demetris said that they are currently focussing on performance with large datasets. The viewer already includes an overview bar, and they included [independent subcoordinate y-axis scaling](https://docs.bokeh.org/en/latest/docs/examples/interaction/tools/subcoordinates_zoom.html) and a [dynamic scale bar](https://docs.bokeh.org/en/latest/docs/examples/basic/annotations/scale_bar_image.html). Here's a short demo video showing off annotations, hover tooltips, and minimap navigation: https://github.com/holoviz-topics/neuro/pull/82

And of course, you can always try out the workflow by installing the conda env described in the main [README](https://github.com/holoviz-topics/neuro/blob/main/README.md).

droumis, 2023-11-17:
Thanks @cbrnr ! 

To reiterate, we welcome any questions or requests.

sappelhoff, 2025-04-17:
Just curious: Is there an update on this / summary of what happened? the last commit on the repo seems to be from 5 months ago, and the initial project time of 12 months should also be over :)

droumis, 2025-04-17:
Sorry for the update delay! We've had to divert attention in multiple directions recently. @cbrnr , @sappelhoff , you can check out the current state of this work here in the HoloViz Gallery: https://examples.holoviz.org/gallery/neuroscience, with the most relevant being [this multichannel timeseries workflow](https://examples.holoviz.org/gallery/multichannel_timeseries/0_multichan.html).

We also have a write-up summary of the project [here](https://blog.holoviz.org/posts/czi_r5/).

We welcome any feedback! We would very much like to continue the conversation of MNE integration.

[image]