mne-tools/mne-python #12230: Replace CircleCI with GitHub Actions

cbrnr, 2023-11-21:
Is there a particular reason why we are still using CircleCI to build our docs? I keep running into issues because I'm not logged in, and they recently disabled the option to simply log in with a GitHub account (or at least they seem to prepare for removing this option).

All of this is to say that it would be much easier if we used GitHub Actions, but maybe there is a reason why this is not possible. If there isn't, would everyone be OK with moving to GHA? And of course, the biggest hurdle is that someone needs to do the migration.

cbrnr, 2023-11-21:
Update: I logged in again (via GitHub), but I still cannot see the logs. There's a message saying "block-unregistered-user". Not great.

[image]

larsoner, 2023-11-21:
1. Load distribution. The more we put on GHA the more we have to wait for runners.
2. We likely have bigger/faster runners on CircleCI because we pay (would need to check)
3. Immediate viewing of rendered docs website. GHA has artifacts but you have to download and unzip yourself. It saves a ton of time as a reviewer to have a clickable link that takes me straight to a rendered browsable website.

There might be more reasons than this but that's what I can think of off the top of my head.

larsoner, 2023-11-21:
> Update: I logged in again (via GitHub), but I still cannot see the logs. There's a message saying "block-unregistered-user". Not great.

That could be because pre-commit CI pushed a commit to your branch, and *it* is not a registered user. If you push a commit to your branch you should be okay.

I'd also recommend doing `pip install pre-commit` and `pre-commit install --install-hooks` to avoid the bot pushing to your fork. Perhaps we should disable the bot pushing since it will always create conflicts with CircleCI

cbrnr, 2023-11-21:
> I'd also recommend doing pip install pre-commit and pre-commit install --install-hooks to avoid the bot pushing to your fork. Perhaps we should disable the bot pushing since it will always create conflicts with CircleCI

I think disabling would be a good idea. People can see what's wrong and push the changes themselves.

But coming back to CircleCI, does this mean that the logs are not publicly accessible (i.e., you have to be logged in)? If so, can we change that and make it public?

larsoner, 2023-11-21:
> does this mean that the logs are not publicly accessible (i.e., you have to be logged in)? If so, can we change that and make it public?

I don't think they are public. But I also don't think logging in via GitHub is too onerous

cbrnr, 2023-11-21:
> I don't think they are public. But I also don't think logging in via GitHub is too onerous

It kind of is. But in addition, we are an open-source project and should develop everything in the open, including our website build logs. If I want to point someone else to some build issue, they need to create an account, log in, and possibly also ask a maintainer to give them permission. That's pretty onerous I think.

So could we make everything public on CircleCI technically?

cbrnr, 2023-11-21:
> ... actually I just looked at https://app.circleci.com/pipelines/github/mne-tools/mne-python/22002/workflows/5c035ee8-e309-4dde-b4f0-13435ad2003b/jobs/61752 in Chrome Incognito and was able to see the logs, so I think they are actually public

Saw that after I posted my comment, I will try!

cbrnr, 2023-11-21:
Yes, they seem to be public. It's just the bot that's problematic. I would either disable auto-fixing or add the bot to our registered users (if that's even possible). WDYT?

larsoner, 2023-11-21:
Let's just disable bot pushing for now. No idea how to add the bot as a registered user

cbrnr, 2023-11-22:
Fixed in #12229.

closed 2023-11-22 by cbrnr