mne-tools/mne-python #12232: Use correct units in PSD plots

cbrnr, 2023-11-22:
OK, nitpicks incoming... but I think we're currently labeling the y-axis in our PSD plots incorrectly. Or at least I'm very confused, so maybe someone can clarify this:

```python
import mne
import numpy as np

fs = 512  # Hz
n_channels = 16
duration = 100  # s
raw = mne.io.RawArray(
    data=np.random.rand(n_channels, duration * fs) * 1e-6,
    info=mne.create_info(n_channels, fs, "eeg"),
)
psd = raw.compute_psd(fmax=60)
psd.plot(dB=True)  # the default
psd.plot(dB=False)
```

This produces two plots:
[image]
[image]

1. The first figure shows the default behavior where `dB=True`, but the y-axis is labeled "µV²/Hz (dB)". This makes no sense, because the unit is either µV²/Hz or dB, but not both.
2. I'd say the term µV²/Hz is mathematically incorrect, it should be either (µV)²/Hz or µ²V²/Hz (the first one seems to be easier on the eyes).
3. The second figure shows the output for `dB=False`, and the y-axis label is "µV/√Hz". I explicitly computed a power spectral density, so why is it showing an amplitude spectral density when `dB=False`? This behavior does not make sense to me.
4. Can we double-check if we compute dB correctly? What is the reference power in the denominator? I think it should be 1 (µV)², which corresponds to 1e-12 V². I gave it a quick shot, and the relevant parts should be [in this function](https://github.com/mne-tools/mne-python/blob/main/mne/viz/utils.py#L2369), but the code is mildly convoluted, so I'd appreciate if someone else could take a look.

I'm really not sure if I completely misunderstand everything, in which case I'd be grateful for any clarifications. Also, it might well be the case that I'm responsible for some (all?) of these inconsistencies, but this doesn't change the fact that I'm confused right now 😝...

Pinging @larsoner and @drammock, but anyone else feel free to chime in!

larsoner, 2023-11-22:
> because the unit is either µV²/Hz or dB, but not both

I think it's a lazy shorthand for `dB re 1uV²/Hz` which as you point out should probably really be `dB re 1(uV)²/Hz`. I'm fine with updating both of these

> The second figure shows the output for dB=False, and the y-axis label is "µV/√Hz". I explicitly computed a power spectral density, so why is it showing an amplitude spectral density when dB=False? This behavior does not make sense to me.

Looks like you brought this up 5 years ago

https://github.com/mne-tools/mne-python/issues/4642#issuecomment-334127918
https://github.com/mne-tools/mne-python/pull/4654

> Can we double-check if we compute dB correctly? What is the reference power in the denominator? I think it should be 1 (µV)², which corresponds to 1e-12 V²

That sounds right to me but agree it would be good to double check

cbrnr, 2023-11-22:
> I think it's a lazy shorthand for dB re 1uV²/Hz which as you point out should probably really be dB re 1(uV)²/Hz. I'm fine with updating both of these

I'd just write dB. I know people like to indicate the reference afterwards, but this looks a bit unwieldy.

> Looks like you brought this up 5 years ago

That's what I suspected, but skimming the issue I was confused by the use of ASD. It seems like this is a MEG thing, but my point is that just changing `dB` from `True` to `False` should not change the PSD to an ASD.

larsoner, 2023-11-22:
> I'd just write dB. I know people like to indicate the reference afterwards, but this looks a bit unwieldy.

To me your suggestion of just using dB is less informative / loses information, so -1 on that from me

cbrnr, 2023-11-22:
> To me your suggestion of just using dB is less informative / loses information, so -1 on that from me

I'm fine with "dB (µV)²Hz⁻¹".

This leaves the other two issues, namely the PSD/ASD confusion and the correctness of calculating dB values.

cbrnr, 2023-11-22:
Since you mentioned consistency with other packages elsewhere, I'd suggest to follow what SciPy does: https://docs.scipy.org/doc/scipy/reference/generated/scipy.signal.periodogram.html#scipy.signal.periodogram

They have a `scaling` parameter, which can either be `"density"` (default) or `"spectrum"`. If you look at the examples, the second one seems to be what you want for MEG – they use `"spectrum"` and the manually compute the square root.

Note that this is completely orthogonal to the `dB` parameter.

hoechenberger, 2023-11-22:
> I'd say the term µV²/Hz is mathematically incorrect, it should be either (µV)²/Hz or µ²V²/Hz (the first one seems to be easier on the eyes).

ChatGPT says:

> When calculating the power of a unit with an SI prefix, like "kg" (kilogram) or "uV" (microvolt), the exponent applies to the entire unit, including the prefix. So, "kg^2" means "(kilogram)^2" and "uV^2" means "(microvolt)^2". This implies that the prefix is also raised to the power of the exponent. For example, "kg^2" is equivalent to "(1000 grams)^2" because "kilo-" means a thousand. Therefore, "kg^2" is equal to "1000000 grams^2" or "a million grams^2". 

Similarly, "uV^2" means "(microvolt)^2", where "micro-" denotes one-millionth. So, "uV^2" is the same as "(10^-6 volts)^2", which equals "10^-12 volts^2". 

In summary, both the base unit and its prefix are affected by the exponent.

cbrnr, 2023-12-01:
Re (1), yes, that was a bit silly. A prefixed unit is treated as one entity, therefore 1 cm³ means 0.001 m³ (= 0.1³ m³). I probably should not overthink everything 😄.

Re (2), OK, sounds good. I prefer "dB µV²/Hz".

Re (3), I do find "µV²Hz⁻¹" a bit more consistent and nicer looking than "µV²/Hz", so my vote goes to "µV²Hz⁻¹" (but also just +0.5).

Re my concern, it's not about dividing by Hz, but the method is called `compute_psd()`, so it should return the PSD, and not the ASD. Even worse, the function currently returns the ASD *by default* if `dB=False`. So what I'm saying is that the `dB` argument should *not* influence if the PSD or ASD is computed.

drammock, 2023-12-01:
> Re my concern, it's not about dividing by Hz, but the method is called `compute_psd()`, so it should return the PSD, and not the ASD. Even worse, the function currently returns the ASD _by default_ [...]

OK, but the `.compute_psd()` methods *do not have parameters `dB` and `amplitude`*. They always return PSDs.  `dB` and `amplitude` are only available in `Spectrum.plot()` or in the legacy `.plot_psd()` methods.

cbrnr, 2023-12-02:
Right, my bad. But I still don't like the fact that `dB` and `amplitude` are not independent.

drammock, 2023-12-03:
> What is the reason we have `amplitude="auto"` here?

did you check `git blame` and/or track down the PR where it was added?

drammock, 2023-12-04:
> I suggest that we disentangle it and remove/deprecate the `amplitude="auto"` setting. The new default should be `dB=True` and `amplitude=False`.

As I said above I'm fine with changing the default of `amplitude` (with depr cycle) from `"auto"` to `False`. I'm less convinced that we need to remove `"auto"` entirely; @larsoner any strong feeling here?

larsoner, 2023-12-04:
If we change the default from auto we might as well remove support for it, it was mostly added at the time for preserving backward compat I think

hoechenberger, 2023-12-05:
+1 for removing it then if it causes trouble

PR #12270 (2023-12-10): Simplify spectrum.plot() parameters
Fixes #12232.

~~I have replaced the `amplitude` parameter with a new `estimate` parameter, which is more intuitive.~~

I am not sure how to change the docstring. Should I include the two parameters? Should I mark `amplitude` as deprecated? If so, how (just mention it or is there some Sphinx role)?

If you want to see the new behavior, here's a small example (I'm filtering unrelated warnings and log messages):

```python
import warnings

from numpy.random import default_rng

from mne import create_info
from mne.io import RawArray
from mne.utils import logger [... cut]
closed 2023-12-10 by drammock