mne-tools/mne-python #12233: DOC: Update changelog sections

larsoner, 2023-11-22:
> We should have na "Dependencies" section in the changelog to mention changes of dependencies. @larsoner WDYT?

_Originally posted by @hoechenberger in https://github.com/mne-tools/mne-python/issues/12218#issuecomment-1822460874_
            
We've had our changelog section names for a long time, but I'm not sure if they're the best. I agree it would be nice to have a place where we mention stuff like the 3.8->3.9 bump, NumPy min changes, and EDFlib-Python -> edfio changes.

I think we should look at how NumPy, SciPy, matplotlib, sklearn, and pandas do things. They've often thought hard about this stuff and it would be nice to maintain consistency with the scientific Python ecosystem.

drammock, 2023-12-15:
Dev meeting discussion landed on this naming and ordering scheme for the changelog sections:

1. dependencies
2. bugfixes
3. API changes
4. new features
5. other changes

@larsoner is planning to adopt this arrangement soon in a PR where we try out towncrier as a changelog management tool.

hoechenberger, 2023-12-15:
I like the section names, but could we reconsider the order please?

I don't think Dependencies need to be at the top because that's not important (as they'll be pulled in automatically when you update), and new features should appear before bugfixes. New features should be at the top to promote them; or after bugfixes if we add a summary of the most significant improvement at the top

1. new features
1. API changes
1. bugfixes
1. dependencies
1. other changes

What would usually go under the last section? Could it make sense to call it Code health? I don't like catch-alls like "other"

drammock, 2023-12-15:
all of your concerns were discussed during the dev meeting. Briefly:

1. dependencies are *not* always pulled in automatically (e.g., for users with a dev install who are updating via `git pull`). And they *are* important for some users, who need to get clearance for new installs from IT.  (aside: entries in the dependencies section should include supplementary info like "only needed if building docs locally" or "needed for Neuralynx file IO" or whatever).
2. "new features" are *not* the most important thing to advertise to users --- bugfixes and API changes can break existing code; new features cannot (by definition). That is why new features is item 4 in my list.
3. we actually *need* a catch-all category; "code health" is too specific. For example, docs-only changes aren't really code health. "other changes" was preferred over "miscellaneous" because "misc" feels a bit minimizing / derogatory.

hoechenberger, 2023-12-15:
Thanks for the summary

I'm not happy with the decision structure in the project. It appears that if one cannot take part in the dev meetings, one is excluded from decision and discussion processes.

This is something we should try to improve upon.

drammock, 2023-12-15:
> Thanks for the summary
> 
> I'm not happy with the decision structure in this project. It appears that if one cannot take part in the dev meetings, one is excluded from decision and discussion processes.
> 
> This is something we should try to improve upon.

I agree 100% that this should be improved (this was in fact also discussed in today's meeting, where we revived the idea of changes to the dev meeting time slot, and I specifically raised concerns about the towncrier PR that it should be presented as "let's try this out as a group" rather than "this is the new way of doing things").

But I disagree with your characterization that

> if one cannot take part in the dev meetings, one is excluded from certain decision and discussion processes.

The whole point of me reporting what was agreed upon at the dev meeting was to *elicit responses* from the people that weren't there. And my follow-up was *a response to the concerns you raised*.  That doesn't seem like exclusion from discussions to me, and in neither of those messages do I say "this is final" (so it's not exclusion from decisions either).  So it's not clear to me what you're objecting to here.  Is it the tone? Something else? Can you clarify?

hoechenberger, 2023-12-15:
Thanks Dan! Seems that was a misunderstanding on my end. I took it as, "the dev meeting has decided and so it shall be done", but this was clearly not what you were intending to communicate. So, sorry about that!

sappelhoff, 2023-12-17:
> *seems this was forgotten for the last release

just a minor note, I deliberately dropped "notable changes" in the last mne-bids release (and perhaps also some other releases that I have done), because I didn't seem any change particularly notable compared to the general list of changes.

I still it's a nice section to (optionally) have :+1:

larsoner, 2023-12-20:
So @hoechenberger can you live with the order/naming in https://github.com/mne-tools/mne-python/issues/12233#issuecomment-1858197294 ?

hoechenberger, 2023-12-20:
Yes let's move ahead

closed 2023-12-20 by larsoner