mne-tools/mne-python #12241: Pylance doesn't know about datasets.sample.data_path()

hoechenberger, 2023-11-25:
### Description of the problem

Pylance claims that `mne.datasets.sample.data_path()` is an unknown entity:
[image]

### Steps to reproduce

```Python
import mne
mne.datasets.sample.data_path()
```

### Expected results

Editor would offer auto-completion and docstrings

### Actual results

Editor doesn't know about `data_path()`

### Additional information

This is on `main`. Problem has existed already before we moved to lazy loading. I tried getting to the bottom of this but I really don't understand what's going on here. No problems at runtime. Applies to all datasets, not only `sample`.

hoechenberger, 2023-11-25:
I'm seeing something similar for `mne.io.read_raw_fif()`, but not for `mne.io.read_raw()`:
[image]

hoechenberger, 2023-11-25:
Can only be reproduced if working in a VS Code workspace that doesn't have the `mne-python` folder open. E.g., do `code /tmp/mwe.py`

cbrnr, 2023-11-25:
As we've just discussed, this works for me... even in a different workspace.

hoechenberger, 2023-11-25:
It seems the problem was in fact related to some type stubs floating around somewhere, so most probably not a bug in MNE or VS Code per se!

closed 2023-11-25 by hoechenberger