mne-tools/mne-python #12255: We may be able to ditch Black in favor of just Ruff

hoechenberger, 2023-12-02:
Ruff has started offering formatting functionality a couple of months back, in addition to just lining. They claim they tested it on large codebases like Django and it yields >99.9% identical results to Black. Differences are e.g. that it would complain about comments exceeding the max. line length, while Black doesn't.

https://docs.astral.sh/ruff/formatter/

In any case, we could theoretically drop Black from our CI and pre-commit hooks and simply use Ruff for everything. This would speed up the pre-commit checks.

WDYT?

hoechenberger, 2023-12-02:
WOW, I just ran `ruff format mne/` and it **immediately** returns, it's blazingly fast! Changes 39 files, though.

Then I ran Black again and it took like 10 seconds or so.

Conclusion: Ruff is just incredibly fast, but formats the occasional line differently than Black. It's more aggressive to keep line lengts below the maximum; e.g., we currently have with Black
```python
            raise ValueError(
                "Number of channels in the Info object (%s) and "
                "the data array (%s) do not match. " % (len(pos["chs"]), data.shape[0])
                + info_help
            )
```

and Ruff formats it as:
```python
            raise ValueError(
                "Number of channels in the Info object (%s) and "
                "the data array (%s) do not match. "
                % (len(pos["chs"]), data.shape[0])
                + info_help
            )
```

drammock, 2023-12-02:
no strong feeling here. Faster is better, for sure, and we should probably switch to Ruff eventually (also eliminates a dependency), but I'd be willing to wait a few more months to let the folks at Astral iron out a few more wrinkles in the Ruff formatter (such as the failure to merge the strings). Which, BTW, you should report upstream if it's not already been reported.

Are the examples you give exhaustive in terms of the kinds of situations that are handled differently?

PR #12261 (2023-12-05): MRG: Use ruff-format instead of Black
Closes #12255

I also removed numerous unnecessary `# noqa: D102`'s from `__init__()` methods. (Unnecessary there if the class itself has a docstring)
closed 2023-12-05 by larsoner