mne-tools/mne-python #12262: API: Use annotations for read_raw_egi events rather than synthetic channel

larsoner, 2023-12-04:
In https://github.com/mne-tools/mne-python/pull/4017#issuecomment-299519890 we added `RawEGI.event_id` which disappears on round-trip. This is due to not knowing which/how many DIN channels will get triggers.

Nowadays we should use Annotations rather than synthesize an event channel I think. Not sure if it's worth keeping the event channel synthesizing around for backward compat or deprecating it. But either way adding the annotations will make it possible for people to use modern code rather than relying on `raw.event_id` here.

See also [discourse](https://mne.discourse.group/t/saving-event-id-attribute-in-raw-object/7932/4) for discussion.

larsoner, 2023-12-04:
Perhaps. But I think switching to mffpy is potentially a much bigger undertaking and so far no one has volunteered. And either way we need to go from DIN events (extracted from somewhere) to `mne.Annotations`, so a first small PR to add DIN-as-annot wouldn't be wasted effort. @scott-huberty if you are interested in learning a bit of EGIMFF the first small PR would be a good place to start :)

scott-huberty, 2023-12-04:
I've worked on that module and have used EGI a lot, so go ahead and assign to me!

PR #12300 (2024-07-16): Use Annotations for read_raw_egi events
Closes #12262 
Closes #11674
Closes #11672

adds a parameter `events_as_annotations` (default False for backward compat), which will create annotations from EGI events.

@larsoner what do you think of this design and the new parameter name?
closed 2024-07-16 by drammock