mne-tools/mne-python #12276: non-interactive brain figure

drammock, 2023-12-07:
### Description of the problem

brain windows are not interactable, don't respond to their close buttons, and crash sometimes when calling `brain.close()`

### Steps to reproduce

```Python
import mne

data_path = mne.datasets.sample.data_path()
subjects_dir = data_path / "subjects"
sample_dir = data_path / "MEG" / "sample"
brain_kwargs = dict(alpha=0.1, background="white", cortex="low_contrast")
brain = mne.viz.Brain("sample", subjects_dir=subjects_dir, **brain_kwargs)
```

### Expected results

an interactive window.

### Actual results

a non-interactable window showing a brain that is zoomed in to the level where it shows about 3 gyri. Closing the window manually doesn't work ("the program isn't responding, wait or terminate?"). Calling `brain.close()` works sometimes, other times it yields this:

```
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
File /opt/mambaforge/envs/mnedev/lib/python3.11/site-packages/pyvistaqt/rwi.py:574, in QVTKRenderWindowInteractor.mousePressEvent(self, ev)
    571 self._ActiveButton = ev.button()
    573 if self._ActiveButton == MouseButton.LeftButton:
--> 574     self._Iren.LeftButtonPressEvent()
    575 elif self._ActiveButton == MouseButton.RightButton:
    576     self._Iren.RightButtonPressEvent()

AttributeError: '_FakeIren' object has no attribute 'LeftButtonPressEvent'
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
File /opt/mambaforge/envs/mnedev/lib/python3.11/site-packages/pyvistaqt/rwi.py:587, in QVTKRenderWindowInteractor.mouseReleaseEvent(self, ev)
    583 self._setEventInformation(pos_x, pos_y,
    584                           ctrl, shift, chr(0), 0, None)
    586 if self._ActiveButton == MouseButton.LeftButton:
--> 587     self._Iren.LeftButtonReleaseEvent()
    588 elif self._ActiveButton == MouseButton.RightButton:
    589     self._Iren.RightButtonReleaseEvent()

AttributeError: '_FakeIren' object has no attribute 'LeftButtonReleaseEvent'
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
File /opt/mambaforge/envs/mnedev/lib/python3.11/site-packages/pyvistaqt/rwi.py:574, in QVTKRenderWindowInteractor.mousePressEvent(self, ev)
    571 self._ActiveButton = ev.button()
    573 if self._ActiveButton == MouseButton.LeftButton:
--> 574     self._Iren.LeftButtonPressEvent()
    575 elif self._ActiveButton == MouseButton.RightButton:
    576     self._Iren.RightButtonPressEvent()

AttributeError: '_FakeIren' object has no attribute 'LeftButtonPressEvent'
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
File /opt/mambaforge/envs/mnedev/lib/python3.11/site-packages/pyvistaqt/rwi.py:587, in QVTKRenderWindowInteractor.mouseReleaseEvent(self, ev)
    583 self._setEventInformation(pos_x, pos_y,
    584                           ctrl, shift, chr(0), 0, None)
    586 if self._ActiveButton == MouseButton.LeftButton:
--> 587     self._Iren.LeftButtonReleaseEvent()
    588 elif self._ActiveButton == MouseButton.RightButton:
    589     self._Iren.RightButtonReleaseEvent()

AttributeError: '_FakeIren' object has no attribute 'LeftButtonReleaseEvent'
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
File /opt/mambaforge/envs/mnedev/lib/python3.11/site-packages/pyvistaqt/rwi.py:576, in QVTKRenderWindowInteractor.mousePressEvent(self, ev)
    574     self._Iren.LeftButtonPressEvent()
    575 elif self._ActiveButton == MouseButton.RightButton:
--> 576     self._Iren.RightButtonPressEvent()
    577 elif self._ActiveButton == MiddleButton:
    578     self._Iren.MiddleButtonPressEvent()

AttributeError: '_FakeIren' object has no attribute 'RightButtonPressEvent'
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
File /opt/mambaforge/envs/mnedev/lib/python3.11/site-packages/pyvistaqt/rwi.py:589, in QVTKRenderWindowInteractor.mouseReleaseEvent(self, ev)
    587     self._Iren.LeftButtonReleaseEvent()
    588 elif self._ActiveButton == MouseButton.RightButton:
--> 589     self._Iren.RightButtonReleaseEvent()
    590 elif self._ActiveButton == MiddleButton:
    591     self._Iren.MiddleButtonReleaseEvent()

AttributeError: '_FakeIren' object has no attribute 'RightButtonReleaseEvent'
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
File /opt/mambaforge/envs/mnedev/lib/python3.11/site-packages/pyvistaqt/rwi.py:574, in QVTKRenderWindowInteractor.mousePressEvent(self, ev)
    571 self._ActiveButton = ev.button()
    573 if self._ActiveButton == MouseButton.LeftButton:
--> 574     self._Iren.LeftButtonPressEvent()
    575 elif self._ActiveButton == MouseButton.RightButton:
    576     self._Iren.RightButtonPressEvent()

AttributeError: '_FakeIren' object has no attribute 'LeftButtonPressEvent'
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
File /opt/mambaforge/envs/mnedev/lib/python3.11/site-packages/pyvistaqt/rwi.py:587, in QVTKRenderWindowInteractor.mouseReleaseEvent(self, ev)
    583 self._setEventInformation(pos_x, pos_y,
    584                           ctrl, shift, chr(0), 0, None)
    586 if self._ActiveButton == MouseButton.LeftButton:
--> 587     self._Iren.LeftButtonReleaseEvent()
    588 elif self._ActiveButton == MouseButton.RightButton:
    589     self._Iren.RightButtonReleaseEvent()

AttributeError: '_FakeIren' object has no attribute 'LeftButtonReleaseEvent'
I
```

### Additional information

```
Platform             Linux-5.15.0-58-generic-x86_64-with-glibc2.35
Python               3.11.6 | packaged by conda-forge | (main, Oct  3 2023, 10:40:35) [GCC 12.3.0]
Executable           /opt/mambaforge/envs/mnedev/bin/python3.11
CPU                  x86_64 (16 cores)
Memory               125.7 GB

Core
├☑ mne               1.7.0.dev29+g4ff2ca41f (devel, latest release is 1.6.0)
├☑ numpy             1.26.0 (OpenBLAS 0.3.24 with 16 threads)
├☑ scipy             1.11.3
├☑ matplotlib        3.8.1 (backend=QtAgg)
├☑ pooch             1.8.0
└☑ jinja2            3.1.2

Numerical (optional)
├☑ sklearn           1.3.2
├☑ numba             0.58.1
├☑ nibabel           5.1.0
├☑ nilearn           0.10.2
├☑ dipy              1.7.0
├☑ openmeeg          2.5.6
├☑ cupy              12.2.0
└☑ pandas            2.1.3

Visualization (optional)
├☑ pyvista           0.42.3 (OpenGL 4.5.0 NVIDIA 470.223.02 via NVIDIA GeForce RTX 2060/PCIe/SSE2)
├☑ pyvistaqt         0.0.0
├☑ vtk               9.2.6
├☑ qtpy              2.4.1 (PyQt5=5.15.8)
├☑ ipympl            0.9.3
├☑ pyqtgraph         0.13.3
├☑ mne-qt-browser    0.7.0.dev2+g2e26973
├☑ ipywidgets        8.1.1
├☑ trame_client      2.12.6
├☑ trame_server      2.12.1
├☑ trame_vtk         2.6.0
└☑ trame_vuetify     2.3.1

Ecosystem (optional)
├☑ mne-bids          0.14.dev0
├☑ mne-nirs          0.6.0
├☑ mne-connectivity  0.6.0dev0
├☑ mne-bids-pipeline 1.5.0.dev42+g1705142
└☐ unavailable       mne-features, mne-icalabel
```

Notice that `pyvistaqt` is marked as version 0.0.0. However:

```console
$ mamba list pyvistaqt
# packages in environment at /opt/mambaforge/envs/mnedev:
#
# Name                    Version                   Build  Channel
pyvistaqt                 0.11.0             pyhd8ed1ab_0    conda-forge
```

larsoner, 2023-12-07:
I can't replicate at least on macOS arm64 with:
```
├☑ pyvista           0.43.dev0 (OpenGL 4.1 Metal - 88 via Apple M1 Pro)
├☑ pyvistaqt         0.12.0.dev55+gee4bbf9
├☑ vtk               9.2.6
├☑ qtpy              2.4.0 (PyQt5=5.15.8)
```
Window shows up fine and I can interact:

[image]

And `brain.close` is okay. I should be able to test on Linux tomorrow.

> Notice that pyvistaqt is marked as version 0.0.0. However:

If you use build 1 instead of build 0 of pyvistaqt 0.11 on conda-forge the version should be fixed, see https://github.com/conda-forge/pyvistaqt-feedstock/pull/18

larsoner, 2023-12-08:
From a quick look at https://app.circleci.com/pipelines/github/mne-tools/mne-python/22194/workflows/a09f0e0f-0303-4214-b67e-14537775d0b6/jobs/62159 which passed and the resulting doc build things are at least not totally broken there. But CircleCI uses PyQt6 I think. Might be worth seeing if install PyQt6 from pip fixes things for you, that would tell us we have some PyQt5 compat problem

larsoner, 2023-12-08:
I can't replicate on Linux with PyQt6 6.6.0 or PyQt5 5.15.10 or 5.15.8 :(

hoechenberger, 2023-12-09:
Works for me on macOS with Qt5

drammock, 2023-12-11:
something seems to be deeply wrong with my system. I can't get it to work with pyqt6 nor pyside6 either:

```
$ QT_API=pyqt6 ipython
In [1]: import mne
   ...: 
   ...: data_path = mne.datasets.sample.data_path()
   ...: subjects_dir = data_path / "subjects"
   ...: sample_dir = data_path / "MEG" / "sample"
   ...: brain_kwargs = dict(alpha=0.1, background="white", cortex="low_contrast")
   ...: brain = mne.viz.Brain("sample", subjects_dir=subjects_dir, **brain_kwargs)
Using pyvistaqt 3d backend.

malloc(): invalid size (unsorted)
Aborted (core dumped)
$ QT_API=pyside6 ipython
In [1]: import mne
   ...: 
   ...: data_path = mne.datasets.sample.data_path()
   ...: subjects_dir = data_path / "subjects"
   ...: sample_dir = data_path / "MEG" / "sample"
   ...: brain_kwargs = dict(alpha=0.1, background="white", cortex="low_contrast")
   ...: brain = mne.viz.Brain("sample", subjects_dir=subjects_dir, **brain_kwargs)
Using pyvistaqt 3d backend.

Installed qt6 event loop hook.
In [2]: Killed
```

with `QT_API=pyside6` I got the same behavior as with `pyqt5`: namely the image was way zoomed in and non-interactive, and the window had to be force-killed as its close button didn't work.

larsoner, 2023-12-11:
:scream: 

Maybe you could try with a Ubuntu-python-based virtualenv instead of conda, just to rule out some badness there? And if that by some miracle works, try a fresh conda env after `conda deactivate`ing your current env, etc.

drammock, 2023-12-11:
doing this:
```console
$ /usr/bin/python3 -m venv mnetest
$ source mnetest/bin/activate
$ pip install mne[dev]
```
followed by running the MWE in the PR description, yields this:
```
RuntimeError: Could not load any valid 3D backend
pyvistaqt: No Qt bindings could be found
notebook: No module named 'ipyevents'

 install pyvistaqt, using pip or conda:
'pip install pyvistaqt'
'conda install -c conda-forge pyvistaqt'

 or install ipywidgets, if using a notebook backend
'pip install ipywidgets'
'conda install -c conda-forge ipywidgets'
```

So our `pip install mne[dev]` apparently doesn't install either pyvistaqt nor ipywidgets?

larsoner, 2023-12-11:
IIRC it installs pyqt but does not specify a Qt binding so you also need for example `pip install PyQt6`

drammock, 2023-12-11:
ok, I've got things working again. Here's what I did:

download our environment.yml file, and remove `pyqt!=5.15.3,!=5.15.4` and replace it with
```
pip:
  - PyQt6
```

a fresh conda env based on that file worked.

closed 2023-12-11 by drammock

hoechenberger, 2023-12-11:
We should probably add ipyevents to the full variant, WDYT?

drammock, 2023-12-11:
> We should probably add ipyevents to the full variant, WDYT?

it's in there, I just made the mistake of installing `pip install mne[dev]` instead of `mne[full,dev]`.  Remember when I said I found it unintuitive/confusing if the `dev` variant didn't install everything?  I wasn't joking 😆

hoechenberger, 2023-12-11:
I find it not great either

Can we not change it?