mne-tools/mne-python #12285: Twitter logo in docs should not be colored

cbrnr, 2023-12-08:
Twitter is the only colored icon on our docs website:

[image]

It should be gray like all the other icons.

drammock, 2023-12-15:
This is inherited from the website theme. In the latest version of the theme the twitter icon color goes away (but that's actually a bug in the theme due to changing the icon from the "square" version to just the bird, but not changing the associated CSS selector). The theme also defines brand-appropriate colors for GitLab and Bitbucket. Github is grey because that's their brand.

In our MNE-Python CSS, we actually define colors for the discord icon:

https://github.com/mne-tools/mne-python/blob/b1329c3ae59d0da3646b0c667441e12ee0f7bd8d/doc/_static/style.css#L9

as well as light and dark variant colors for discourse, github, and mastadon icons, e.g.:

https://github.com/mne-tools/mne-python/blob/b1329c3ae59d0da3646b0c667441e12ee0f7bd8d/doc/_static/style.css#L46-L48

but for some reason they aren't getting used, so that's also a bug.

Finally, one might argue we should change the twitter logo to an X logo (but that should probably happen at the theme level).

With all that said: do you still think they should be all grey, or should we try to fix the broken color display of the other icons?

cbrnr, 2023-12-16:
Thanks for investigating @drammock! Either way is fine with me, although I have a slight preference for using color.

PR #12301 (2023-12-18): fix icon link colors
closes #12285
closed 2023-12-18 by larsoner